motioneye-project/motioneyeos #1615: MJPEG camera advanced parameters not working anymore...

Kywalh, 2018-08-09:
Hello, 
I added a network camera recently and struggled with the latency of the new one.
As a consequence, I deleted the one I had **and that was working very fine before**, a simple Foscam MJPEG camera.

Now, when I add it back, I cannot do anything I was doing before... motion detection, text overlay, etc...
I can only change the camera name, nothing else...

[image]

Note that when I was struggling with the network camera, I activated the FNC but it is deactivated now...

Any idea ?

### motionEyeOS Version

I am running motionEyeOS version: 
motionEye Version	0.39.2
Motion Version	4.1.1
OS Version	motionEyeOS 20180627

### Board Model

I am using the following board/model: Raspberry PI 3B+

### Camera

I am using the following type of camera: one Network Camera + one Simple MJPEG Camera.

My MJPEG camera model is: FOSCAM FI8918W
My Network camera model is: Szsinocam SN-IPC-HW16

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:
/NA

### Log Files

I consider the following log files relevant to this issue:

[dmesg.log](https://github.com/ccrisan/motioneyeos/files/2275123/dmesg.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/2275124/motioneye.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/2275125/messages.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/2275126/boot.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/2275127/motion.log)

ccrisan, 2018-08-09:
Add it as a Network Camera, not as a Simple MJPEG Camera.

closed 2018-08-09 by ccrisan

Kywalh, 2018-08-09:
Problem solved ! Thx