motioneye-project/motioneyeos #1619: System crashes within minutes

win98se, 2018-08-13:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: **dev20180730**

### Board Model

I am using the following board/model: **Raspberry Pi Zero W**

### Camera

I am using the following type of camera: **MMAL**

My camera model is: **Raspberry Pi Camera Module v1.3**

### Network Connection

My motionEyeOS unit is connected to the network via: **WiFi**

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * **None**

### Log Files

I consider the following log files relevant to this issue:

 * [motion.log](https://github.com/ccrisan/motioneyeos/files/2281851/motion.log)
 * [motioneye.log](https://github.com/ccrisan/motioneyeos/files/2281852/motioneye.log)
 * [messages.log](https://github.com/ccrisan/motioneyeos/files/2281854/messages.log)
 * [boot.log](https://github.com/ccrisan/motioneyeos/files/2281855/boot.log)
 * [dmesg.log](https://github.com/ccrisan/motioneyeos/files/2281856/dmesg.log)
 * [motion.conf](https://github.com/ccrisan/motioneyeos/files/2281858/motion.conf.txt)
 * [thread-1.conf](https://github.com/ccrisan/motioneyeos/files/2281859/thread-1.conf.txt)

The system always crash after a several minutes since boot. I did not attach any HDMI or USB devices.

laguten, 2018-08-14:
I have the same problem in exactly the same configuration with the versions motionEyeOS 20180224, 20180627 and 20180602.  ((

McMornan, 2018-08-19:
Hi, I have the same issue. I used to work with a version from last June. After doing an update to all my motioneye cameras, I get 30 crashes a day - rendering the cameras unusable. I can reproduce it on raspi zero, raspi zero W, raspi 3 and Raspi 3b. I tried freshinstalls from all versions since 1st of January. (wiped microcard (4 different), did a fresh install. ) Is there a way to increase debug level ? I can provide debug logs, currently the log in one of the raspis looks like this, which doos not show any error I can identify:

(its a brand new raspi zero w with brand new sd card and official raspi power unit)

Aug 19 17:55:15 meye-0c9b43e5 daemon.info : starting pid 1055, tty '/dev/ttylogin': '/sbin/getty -L ttylogin 115200 vt100'
Jan  1 00:00:08 meye-0c9b43e5 daemon.info dhclient: DHCPREQUEST on wlan0 to 255.255.255.255 port 67
Jan  1 00:00:08 meye-0c9b43e5 daemon.info dhclient: DHCPACK from 192.168.2.1
Jan  1 00:00:09 meye-0c9b43e5 daemon.info dhclient: bound to 192.168.2.143 -- renewal in 904402 seconds.
Aug 19 18:01:09 meye-0c9b43e5 cron.info crond[822]: crond (busybox 1.27.2) started, log level 8
Aug 19 18:01:09 meye-0c9b43e5 cron.info crond[825]: crond (busybox 1.27.2) started, log level 8
Aug 19 18:01:09 meye-0c9b43e5 daemon.info dhclient: DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 8
Aug 19 18:01:09 meye-0c9b43e5 daemon.info dhclient: DHCPREQUEST on wlan0 to 255.255.255.255 port 67
Aug 19 18:01:09 meye-0c9b43e5 daemon.info dhclient: DHCPOFFER from 192.168.2.1
Aug 19 18:01:09 meye-0c9b43e5 daemon.info dhclient: DHCPACK from 192.168.2.1
Aug 19 18:01:09 meye-0c9b43e5 auth.info sshd[851]: Server listening on 0.0.0.0 port 22.
Aug 19 18:01:09 meye-0c9b43e5 daemon.info dhclient: bound to 192.168.2.143 -- renewal in 834816 seconds.
Aug 19 18:01:10 meye-0c9b43e5 daemon.notice proftpd[869]: localhost - ProFTPD 1.3.6 (stable) (built Tue Jan 2 2018 02:30:31 EET) standalone mode STARTUP
Aug 19 18:01:20 meye-0c9b43e5 user.notice motion: [-1226673552:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
Aug 19 18:01:20 meye-0c9b43e5 user.alert motion: [-1226673552:motion] [ALR] [ALL] conf_cmdparse: Deprecated config option "thread" since after version 3.4.1:
Aug 19 18:01:20 meye-0c9b43e5 user.alert motion: [-1226673552:motion] [ALR] [ALL] conf_cmdparse: The "thread" option has been replaced by the "camera" option.
Aug 19 18:01:20 meye-0c9b43e5 user.notice motion: [-1226673552:motion] [NTC] [ALL] config_camera: Processing camera config file thread-1.conf
Aug 19 18:01:20 meye-0c9b43e5 user.notice motion: [-1226673552:motion] [NTC] [ALL] motion_startup: Motion 4.1.1 Started
Aug 19 18:01:20 meye-0c9b43e5 user.notice motion: [-1226673552:motion] [NTC] [ALL] motion_startup: Logging to syslog
Aug 19 18:01:20 meye-0c9b43e5 user.notice motion: [-1226673552:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
Aug 19 18:01:20 meye-0c9b43e5 user.notice motion: [-1226673552:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Aug 19 18:01:26 meye-0c9b43e5 daemon.info : starting pid 1050, tty '/dev/ttylogin': '/sbin/getty -L ttylogin 115200 vt100'
Jan  1 00:00:10 meye-0c9b43e5 daemon.info dhclient: DHCPREQUEST on wlan0 to 255.255.255.255 port 67
Jan  1 00:00:10 meye-0c9b43e5 daemon.info dhclient: DHCPACK from 192.168.2.1
Jan  1 00:00:10 meye-0c9b43e5 daemon.info dhclient: bound to 192.168.2.143 -- renewal in 681326 seconds.
Aug 19 18:09:23 meye-0c9b43e5 cron.info crond[827]: crond (busybox 1.27.2) started, log level 8
Aug 19 18:09:23 meye-0c9b43e5 cron.info crond[830]: crond (busybox 1.27.2) started, log level 8
Aug 19 18:09:23 meye-0c9b43e5 daemon.info dhclient: DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 3
Aug 19 18:09:23 meye-0c9b43e5 daemon.info dhclient: DHCPREQUEST on wlan0 to 255.255.255.255 port 67
Aug 19 18:09:23 meye-0c9b43e5 daemon.info dhclient: DHCPOFFER from 192.168.2.1
Aug 19 18:09:23 meye-0c9b43e5 daemon.info dhclient: DHCPACK from 192.168.2.1
Aug 19 18:09:23 meye-0c9b43e5 auth.info sshd[854]: Server listening on 0.0.0.0 port 22.
Aug 19 18:09:23 meye-0c9b43e5 daemon.info dhclient: bound to 192.168.2.143 -- renewal in 842030 seconds.
Aug 19 18:09:24 meye-0c9b43e5 daemon.notice proftpd[873]: localhost - ProFTPD 1.3.6 (stable) (built Tue Jan 2 2018 02:30:31 EET) standalone mode STARTUP
Aug 19 18:09:34 meye-0c9b43e5 user.notice motion: [-1227197840:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
Aug 19 18:09:34 meye-0c9b43e5 user.alert motion: [-1227197840:motion] [ALR] [ALL] conf_cmdparse: Deprecated config option "thread" since after version 3.4.1:
Aug 19 18:09:34 meye-0c9b43e5 user.alert motion: [-1227197840:motion] [ALR] [ALL] conf_cmdparse: The "thread" option has been replaced by the "camera" option.
Aug 19 18:09:34 meye-0c9b43e5 user.notice motion: [-1227197840:motion] [NTC] [ALL] config_camera: Processing camera config file thread-1.conf
Aug 19 18:09:34 meye-0c9b43e5 user.notice motion: [-1227197840:motion] [NTC] [ALL] motion_startup: Motion 4.1.1 Started
Aug 19 18:09:34 meye-0c9b43e5 user.notice motion: [-1227197840:motion] [NTC] [ALL] motion_startup: Logging to syslog
Aug 19 18:09:34 meye-0c9b43e5 user.notice motion: [-1227197840:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
Aug 19 18:09:34 meye-0c9b43e5 user.notice motion: [-1227197840:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[... cut ...]
Aug 19 18:35:47 meye-0c9b43e5 user.notice motion: [-1226595728:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Aug 19 18:35:53 meye-0c9b43e5 daemon.info : starting pid 1050, tty '/dev/ttylogin': '/sbin/getty -L ttylogin 115200 vt100'
Aug 19 18:50:13 meye-0c9b43e5 auth.info sshd[2922]: WARNING: could not open /data/etc/moduli (No such file or directory), using fixed modulus
Aug 19 18:50:16 meye-0c9b43e5 auth.info sshd[2922]: Accepted none for admin from 192.168.2.141 port 64970 ssh2

ccrisan, 2018-08-19:
@McMornan can you please describe in more detail what your problem is? Is the system still reachable via SSH when it crashes? You can enable debugging in Expert Settings for slightly more information in your logs.