motioneye-project/motioneyeos #1620: OPI One: cameras not detected

sambul13, 2018-08-13:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180627).

### Board Model

I am using the following board/model: OrangePI PC
### Camera

I am using the following type of camera: V4L2
My camera model is: Logitech C910, also Logitech QuickCam Communicate STX
### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

ccrisan, 2018-08-13:
Please attach a screenshot and make sure you log in with admin.

sambul13, 2018-08-13:
Sorry, forget to switch on Advanced Settings in browser. No cams detected - see logs.
[MotionEyeOSLogs.zip](https://github.com/ccrisan/motioneyeos/files/2282829/MotionEyeOSLogs.zip)

sambul13, 2018-08-13:
I reconnected the cam to another USB port:

[Thu Jan  1 00:40:09 1970] usb 5-1: new high-speed USB device number 3 using ehci-platform
[Thu Jan  1 00:40:09 1970] usb 5-1: New USB device found, idVendor=046d, idProduct=0821
[Thu Jan  1 00:40:09 1970] usb 5-1: New USB device strings: Mfr=0, Product=0, SerialNumber=1
[Thu Jan  1 00:40:09 1970] usb 5-1: SerialNumber: 14C5C1F0
[Thu Jan  1 00:40:10 1970] uvcvideo: Found UVC 1.00 device <unnamed> (046d:0821)
[Thu Jan  1 00:40:10 1970] uvcvideo 5-1:1.2: Entity type for entity Extension 4 was not initialized!
[Thu Jan  1 00:40:10 1970] uvcvideo 5-1:1.2: Entity type for entity Processing 2 was not initialized!
[Thu Jan  1 00:40:10 1970] uvcvideo 5-1:1.2: Entity type for entity Camera 1 was not initialized!
[Thu Jan  1 00:40:10 1970] uvcvideo 5-1:1.2: Entity type for entity Extension 5 was not initialized!
[Thu Jan  1 00:40:10 1970] uvcvideo 5-1:1.2: Entity type for entity Extension 6 was not initialized!
[Thu Jan  1 00:40:10 1970] uvcvideo 5-1:1.2: Entity type for entity Extension 7 was not initialized!
[Thu Jan  1 00:40:10 1970] uvcvideo 5-1:1.2: Entity type for entity Extension 8 was not initialized!
[Thu Jan  1 00:40:10 1970] input: UVC Camera (046d:0821) as /devices/platform/soc/1c1b000.usb/usb5/5-1/5-1:1.2/input/input3
[Thu Jan  1 00:40:10 1970] usbcore: registered new interface driver uvcvideo
[Thu Jan  1 00:40:10 1970] USB Video Class driver (1.1.1)

sambul13, 2018-08-13:
However, only one cam works at a time, and only recognized when connected to the lower USB port. I then removed the 1st cam, connected the 2nd cam to the same lower USB port, and it got recognized. The 1st cam settings still remain in MotionEye browser view, but the cam box is empty, doesn't show video. And dmesg log shows its not recognized upon reboot. So, why only one cam at a time is working?

ccrisan, 2018-08-13:
A power problem or a USB problem maybe?

sambul13, 2018-08-13:
Other USB ports work too simultaneously with other devices like Mouse etc, but don't recognize cams. Its not a power problem either, since if just one cam is hooked to another USB port, and no other devices, its still don't recognized, only in port 5-1. Something is not enabled in OS?

ccrisan, 2018-08-13:
Then I have no idea.

sambul13, 2018-08-13:
I tested the same board with [Armbian Next](https://dl.armbian.com/orangepipc/nightly/), and the same cams are recognized on each USB port. May be its a kernel & module & config issue?

ccrisan, 2018-08-13:
I'll update the kernel of all boards for the next version.

sambul13, 2018-09-02:
Hi ccrisan,

When will you have a chance to update the kernel?

ccrisan, 2018-09-02:
Not anytime soon, as I'm busy with other things.

mdPlusPlus, 2019-11-08:
Has this been resolved yet?