motioneye-project/motioneyeos #1622: Authentication failed when adding network camera

ivodi, 2018-08-14:
Hello, I have just started using MotionEye - great work indeed. 

I am having issues adding a simple network camera using the url that is used for streaming to a browser

http://192.168.89.240/videostream.cgi

[image]

And I am always getting the authentication failed message despite being sure that I am not making mistakes in typing the password:
[image]

How can I find a fix for this issue? 

Thank you

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20180730

### Board Model

I am using the following board/model: Raspberry PI 3B+

### Camera

I am using the following type of camera: Network Camera 

My camera model is: Inspector IIPC-20.

### Network Connection

WiFi

### Peripherals

None

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
2018-08-14 14:28:14: [motioneye]     INFO: hello! this is motionEye server 0.39.3
2018-08-14 14:28:15: [motioneye]  WARNING: camera config file at /data/etc/thread-2.conf is incomplete, ignoring
2018-08-14 14:28:17: [motioneye]     INFO: cleanup started
2018-08-14 14:28:17: [motioneye]     INFO: wsswitch started
2018-08-14 14:28:17: [motioneye]     INFO: tasks started
2018-08-14 14:28:17: [motioneye]     INFO: mjpg client garbage collector started
2018-08-14 14:28:17: [motioneye]     INFO: smb mounts started
2018-08-14 14:28:17: [motioneye]     INFO: server started

ccrisan, 2018-08-21:
@ivodi assuming you have entered your credentials correctly, this might be due to your camera using Digest Authentication instead of Basic Authentication. Reading from Motion documentation page:

    "The Username and password for the network camera. For http protocols, this option is for HTTP 1.1 Basic authentication only. The string is specified as username:password. To use no authentication simply remove this option. Digest authentication is only available for rtsp/rtmp cameras."

closed 2018-08-21 by ccrisan

tokamac, 2019-05-03:
Same problem here. Hassbian Strech, installed motion 4.2.2 fine, then motioneye with pip2 (to avoid issues with Python3). Worked like a charm until I installed wrong `/dev/video*` for my camera. Was immediately unlogged, and from this moment on I can never log in MotionEye GUI. When I type my ID and password I get "invalid credentials". If I enter again my password, a new blank window opens with the address `http://MYI_P/login/`
I uninstalled and reinstalled, rebooted in each time… motioneye is totally broken. 
Any solution? For example can we erase the file containing that password? Where is it located?