motioneye-project/motioneyeos #1625: Testing MotionEyeOS on Pi 1B without networking

Viewfinder98, 2018-08-18:
I'm looking to test MotionEyeOS on my old Pi 1 Model B before buying Pi 3 B+.
I've created the SDHC card install using the Etcher method and it starts the boot process before restarting due to no network.

I'm hoping to run this standalone locally without networking and I've searched issues page here and other forums but can't find any solutions for this model apart from mentions of os.conf file variable os_networkless="true" which doesn't seem to exist in the SD card image I created using the Etcher method.

For the purposes of testing MotionEye, is there a way to skip network dependencies if I don't have /data/etc/os.conf?

The contents of SDHC card:

bcm2708-rpi-0-w.dtb
bcm2708-rpi-b.dtb
bcm2708-rpi-b-plus.dtb
bcm2708-rpi-cm.dtb
bootcode.bin
cmdline.txt
config.txt
fixup.dat
fwupdater.gz
kernel.img
start.elf

I am running motionEyeOS version: motioneyeos-raspberrypi-20180627.img
Raspberry Pi 1 Model B
Hoping to try various USB webcams, initially MS Lifecam

Thanks.
Apologies if Issues is not the place for support questions.

ccrisan, 2018-08-21:
For now you'll have to create your os.conf manually. One way is to boot it up with network and it will create the file automatically. Then you can edit it and set `os_networkless` to `true`.

Viewfinder98, 2018-08-22:
@ccrisan Thanks! Will give that a try.
I did try and make the os.conf file manually under the correct /data/etc structure the other day which didn't make a difference but will do as you suggest and let it create it automatically this time.