motioneye-project/motioneyeos #1626: 1920x1400 custom resolution and brightness

0KiB, 2018-08-21:
ISSUE: If running resolution W1920 x H1400 brightness wont work correct, after 10minutes camera goes white because of sun-light changes and wont recover. 1920x1080 resolution works good but i cant get advantage from fisheye lens so view area isnt good enought. I saw similar issue at motioneye github that chmod /motion/ folder should fix that but its not a solution for motioneyeos(tested / not working). Thanks. Apologize bad english

### motionEyeOS Version

20180627

### Board Model

Raspberry PI 3B.

### Camera

V4L2
Ebay 5MP 1080p 160Fisheye

### Network Connection

Ethernet.
-------

jasaw, 2018-08-23:
Try editing  `/etc/modules`, change `bcm2835-v4l2 max_video_width=1920 max_video_height=1200` to whichever resolution you need, like `bcm2835-v4l2 max_video_width=1920 max_video_height=1400`.

Higher resolution uses more GPU memory, so make sure you have allocated enough memory for the GPU.

jasaw, 2018-09-30:
Does auto-brightness work with this zoomed in view? If yes, you're heading the right direction. Try playing with the `max_video_width` and `max_video_height` numbers, change them to this to start with: `max_video_width=2592 max_video_height=1944`

Auto try playing with the resolution in MotionEye. I suspect some aspect ratios might work better than others.

0KiB, 2018-11-24:
> Does auto-brightness work with this zoomed in view? If yes, you're heading the right direction. Try playing with the `max_video_width` and `max_video_height` numbers, change them to this to start [...]

I was unable to get it working at all by changing different resolutions. But when I set it to fast network camera mode it works perfect. I wonder why its not working as normal mode.