motioneye-project/motioneyeos #1627: Cannot capture still images (no option here) or upload pictures to NAS

Hack4Life, 2018-08-23:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20180627

### Board Model

I am using the following board/model: Raspberry Pi Zero W

### Camera

I am using the following type of camera: Fast Network Camera with Camera Type Simple MJPEG Camera

My camera model is: OV5647 .

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

-

### Log Files

I consider the following log files relevant to this issue:

-

### Problem
I want to setup the camera to take pictures when motion is detected and also upload the picture to my NAS. When I log in as admin, there is no option where I can do this. Here is a view of what I see when I'm logged in: [click for image](https://www2.pic-upload.de/img/35839365/2018_08_23-MotionEyeOS.png).

ccrisan, 2018-08-23:
Enable General Settings > Advanced Settings

closed 2018-08-23 by ccrisan

Hack4Life, 2018-08-24:
If you look at the Screenshot I linked you can see that Advanced Settings are enabled.

ccrisan, 2018-08-24:
I'm sorry but i can't tell that from the image you sent me.

cecilo007, 2018-10-11:
> ### Preliminary Docs
> I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.
>
> I confirm that I have [...]

Hi,
i just solved this problem. After few hours of trying.  Switch off the fast ethernet camera option in expert settings. Simple, but works. 

bye t