motioneye-project/motioneyeos #1629: How to: wpa_supplicant.conf on Mac

Cerberus2, 2018-08-23:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am trying to run motionEyeOS version:  20180627 on a Raspberry Pi Zero W

Raspberry Pi v2 NOIR Camera

I am trying to run motioneyeos on a pi zero w. I have successfully  downloaded and install the motioneyeos  img file to my SD card using Etcher. The problem is that I am having trouble connecting to the wifi, I have wrote and uploaded the wpa_suppliant.conf to the SD card as well. Once inserting to SD card into the pi zero w, the pi never is able to connect to the wifi. I have completed all the setups using a OSX, windows, and linux, still the pi will not generate an IP address or connect to the wifi.

I am looking for possible solutions?

Thanks!

jasaw, 2018-08-24:
Have you read through the troubleshooting guide?
https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#wireless-connection-problems

Boksy, 2018-08-26:
I believe it only reads the file on first boot.