motioneye-project/motioneyeos #1630: Pi zero the image does not exit

tigernero79, 2018-08-23:
with version 30072018 known that with "pi zero" the image does not exit, instead it exits a prohibition symbol, and rarely the gui is not available but only access ssh, from ssh if I click reboot nothing happens, the "pi zero" is not restarted . what are the commands to use via the shell?

ccrisan, 2018-08-26:
@tigernero79 I don't understand what this issue is about. Please clearly state what you're trying to do, what you expect the system to do and what it does instead.

tigernero79, 2018-08-27:
the problem of the lack of image in the motioneyeos gui rarely happens, perhaps in conjunction with current dips, and maybe even for the same reason I have access only to the terminal via SSH and not via the web. although the reboot command does not work in this case. I speak after falling electricity

closed 2018-08-27 by tigernero79