motioneye-project/motioneyeos #1636: Raspberry Pi 3B+ AND POE-HAT Support?

kawakawa77, 2018-08-26:
I am running motionEyeOS version: 20180627

I am using the following board/model: Raspberry PI 3B+

I am using the following type of camera: Raspberry Pi Spy Cam (vc.ril.camera / MMAL Camera)

Hello,

here comes the problem...

I configured MotionEyeOs to work als WIFI-Device. It worked fine, I got video from attached camera.

Today I got the new POE-HAT for Raspi 3B+ and installed it.

While MotionEyeOs was working on WIFI I accessed the Web-Interface and disabled WIFI. I have all network credentials defined as static (no DHCP).

I disconnected the power supply from my PI.

I inserted the network cable into my Pi, where the other end was connected to a POE-capable switch.

The Pi started. I was able to connect to the Web-Interface of MotionEyeOs.

The only difference was, that the camera LED was NOT lighting. And I got no video from camera in the Web-Interface. I am getting the picture, which is showing a stroken through camera only.

Then I shutdown the Pi by Web-Interface an disabled the Pi by pulling the network cable from it.

I startet with network cable inserted again, but there was no difference.

So I enabled WIFI in the Web-Interface of the Pi again and restarted it. The Pi was running in WIFI-Mode now.

After connecting to the Web-Interface, I saw the camera video and the camera LED was lighting.

So where could be the problem? When the Pi is supported with power by the POE-HAT, I am getting no video signal from camera or the PI does not recognise the attached camera. While the Pi is powered by the standard power supply, all things are fine.

Do you have an idea, where is the problem?

Does the POE-HAT need an future update of MotionEyeOs? Is the POE-HAT not supported yet?

Thank you for your answers.

Regards
Artur

P.S.
Excuse my bad english.

bluedavid123, 2018-08-26:
im using a pi 3b+ with poe hat, everything works fine, except there is a Raspbian update for the poe hat that enables the cooling fan to operate, how do I get this update under motioneyesos? as my pi is running very hot to the point of shutdown

ccrisan, 2018-08-26:
@kawakawa77 I'm going to release a new version of motionEyeOS that includes updates to various components (such as Kernel and Firmware). However, I am not planning on adding support for various hats in motionEyeOS.

If the update won't solve your problem, you'll have two options:

* use Raspbian + motionEye
* manually add required device tree files to your boot partition (after each update)

kawakawa77, 2018-08-26:
@ccrisan Thank you for your answer.

The PoE-HAT I mean, is not any PoE-HAT from "some producer XY".

I mean the *official* PoE-HAT.

This one: https://www.raspberrypi.org/products/poe-hat/

I understood the politics by releasing the Raspi 3B+ model, that the users will get a official PoE powering option. So the Raspi could be used in environments, where is no electricity available, but a network cable.

Maybe you can overthink this matter.

ccrisan, 2018-08-28:
I did not know we now have an official hat for PoE. Thanks for clarifying that for me. We could add some more DTBs to the OS image - I'll have to think about it and see what criteria to use for selecting them.

kawakawa77, 2018-08-28:
Thank you very much.

mafli77, 2018-09-20:
This feature would be awesome! The official PoE hat is really great for building CCTV cams. Thus it would be great if it could be supported by motioneyeos.

mafli77, 2018-09-25:
The fan of the POE hat is controlled via I2C via the raspi‘s firmware mailbox property.

The original raspi operating system powers the fan according to the temperature of the CPU.

Maybe the following information from another project which uses an own operating system are helpful for implementing this feature in motioneyeos:

Analysis:
https://lore.kernel.org/patchwork/cover/980275/

Implementation:
https://lore.kernel.org/patchwork/patch/980273/
https://lore.kernel.org/patchwork/patch/980274/

raquintanilla, 2019-02-09:
Is there any update on using the PoE hat?

ZipperZ, 2019-03-19:
Maybe someone has made the POE hat fan to work for them? If so can you share your solution?

0emanresuym, 2019-04-01:
We can do this. Let's get the Official PoE hat FAN working for motioneyeos.

This is a good link to something similar:
https://jumpnowtek.com/rpi/Compiling-Raspberry-Pi-Overlays-with-Buildroot.html

I am trying to figure out how to "build the kernel" because I already did what is mentioned in this link:
https://raw.githubusercontent.com/raspberrypi/firmware/master/boot/overlays/README

which was to add to /boot/config.txt the following line:
dtoverlay=rpi-poe

I rebooted a couple times and the fan is still not working.

So, we need to "build the kernel" or something that will actually enable the PoE Hat FAN.

Fuoji, 2019-07-02:
Want to bump this thread for more support. Think this would be a great addition to Motion. I have multiple cameras running off Motion and basically just wired the fan to my header pins because it really gets up there in temp. The downside being I have no control over the fan and it just runs constantly...I forsee problems with that.

TokenChingy, 2019-08-28:
Any updates to this?

onnozweers, 2019-09-29:
Any updates yet? I built a Pi 3B+ plus infrared cam plus the PoE hat, it works great, except for the dead fan:

```
[root@meye-a650e981 ~]# vcgencmd measure_temp
temp=69.8'C
```

I fear this limits the lifetime of the hardware.

ccrisan, 2019-09-29:
If everything goes well, the next nightly build will include support for device-tree overlays. More specifically, the [dtoverlays](https://github.com/ccrisan/thingos/wiki/dtoverlays) configuration file can be used to load additional device tree overlays from `/boot/overlays`. All overlays available for the Raspberry Pi kernel are now provided by motionEye as well.

I'm hoping this is the solution to this issue.

onnozweers, 2019-10-05:
I installed dev20191002 (https://github.com/ccrisan/motioneyeos/releases/tag/nightly-dev) and I'm happy to confirm the fan now works. The temperature nicely stays around 50°C.

```
[root@egelcam1 ~]# vcgencmd measure_temp
temp=50.5'C
[root@egelcam1 ~]# vcgencmd measure_temp
temp=49.9'C
[root@egelcam1 ~]# vcgencmd measure_temp
temp=49.9'C
[root@egelcam1 ~]# vcgencmd measure_temp
temp=50.5'C
```

However, it turns out the fan makes quite a bit of noise. My cam is in a hedgehog house, and I'm pretty sure the noise will scare away any potential residents. So I will switch off the fan and try reducing frame rate and resolution to keep the RPi3 cool.

But nonetheless, thanks a lot for implementing this!

ccrisan, 2019-10-05:
@onnozweers thanks for getting back to us!

closed 2019-10-05 by ccrisan

onnozweers, 2019-10-05:
You're welcome!

By the way, in case anyone (after upgrading) wants to turn off the fan: ssh as admin into the OS, then

```
mount /boot -o remount,rw
```

Then add this line to `/boot/config.txt`:

```
disable_poe_fan=1
```

Then reboot.

ildylan, 2019-10-12:
Installed just now.
Fan. Is. SPINNING! 
Thanks for doing this everyone.