motioneye-project/motioneyeos #1640: Email Motion notifications via gmail

srjjohnson, 2018-08-28:
I started using motioneyeos on raspberry pi and is very easy to install and use. I recently decided to setup email alerts when motion detected.

email address      AAAAA@gmail.com
smtp server          smtp.gmail.com
smtp port             587
smtp account        AAAAA@gmail.com
smtp password     XXXXXXX
from address        AAAAA@gmail.com
use tls                  ON
attach pictures time span   5

I also set gmail to allow for "Less secure apps".
https://accounts.google.com/DisplayUnlockCaptcha

I click on "TEST EMAIL" and nothing happens. No feedback from the app and no email received.

### motionEyeOS Version

motioneyeos version 20180602 
motion version 4.1.1 
motioneye 0.39.1

### Board Model

Raspberry Pi Zero W

### Camera

Raspberry pi camera V2 from Element14.com

### Network Connection
Wifi

### Peripherals
None

### Log Files

I consider the following log files relevant to this issue.

[motioneye.log](https://github.com/ccrisan/motioneyeos/files/2329444/motioneye.log)

ccrisan, 2018-08-28:
@srjjohnson your log indicates that you have a network misconfiguration issue:

      1970-01-01 08:11:36: [sendmail]    ERROR: failed to send mail: [Errno -2] Name or service not known
      Traceback (most recent call last):
      ...
      gaierror: [Errno -2] Name or service not known

More precisely, your DNS is not working. Please make sure your network connection is fully functional and then your gmail configuration will also work.

srjjohnson, 2018-08-28:
Everything seems to be working correctly. Is there somehow I can test to verify?

srjjohnson, 2018-08-29:
Before I ran from my laptop. I ran from chrome browser on raspberry pi 3 and received a message

"Notification email failed: check SMTP server name!"

I've attached a file that may be helpful? 
[motioneye (3).log](https://github.com/ccrisan/motioneyeos/files/2330405/motioneye.3.log)

srjjohnson, 2018-08-29:
Thank you for the tip. I was able to get it working.

closed 2018-08-29 by srjjohnson

maltaresk, 2020-03-08:
i m getting username password not accepted. thought its correct

maltaresk, 2020-03-08:
Send An Email | ON |  
Email Addresses |  gmail  
SMTP Server |   smtp.gmail.com 
SMTP Port |   | 587
SMTP Account |   gmail
SMTP Password |   gmail pswd
From Address |   gmail 
Use TLS | ON |  
Attached Pictures Time Span | seconds | 5 sec 
  | Test Email |

GuilleTCast, 2020-05-17:
In google accounts maybe you should allow less secure apps

starbasessd, 2020-05-17:
Email Address: <whatever>@gmail.com
SMTP Port: 587
SMTP Account: <whatever>    (without the @gmail.com)
From Address:  <whatever@gmail.com
Use TLS=On
Always click Apply before testing.
[image: image.png]

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Bazmundi, 2021-12-18:
I worked out that you need to go to google account and create a one-time app password.   This is because google have beefed up their security.