motioneye-project/motioneyeos #1645: wpa_supplicant.conf for a WIFI requesting both username and password

robertofabrizi, 2018-09-03:
### motionEyeOS Version

I am running motionEyeOS version: 20180627

### Board Model

I am using the following board/model: PI ZERO W

### Camera

Not relevant.

### Network Connection

WIFI

### Peripherals

Not relevant.

### Log Files

Not available. 

If I use the wpa_supplicant found in the docs I get a wpa_supplicant: failed (brcmfmac) when connecting to a wifi that requires both a username and password. How can I specify both? Also, since the password expires every 3 months, how can I change it? I've read that the wpa_supplicant is read just once at the first boot, so I understand that changing it there won't help?

Thanks everyone,
Roberto

tigernero79, 2018-09-04:
Il tuo problema è che usi autentificazione wpa enterprise. Hai a monte qlc firewall di rete tipo ipsense o zeroshell con autentificazione Radius.

Per impostare Il wpa supplicant

Usa questa voce:

network={
  ssid="YOUR_SSID"
  scan_ssid=1
  key_mgmt=WPA-EAP
  identity="YOUR_USERNAME"
  password="YOUR_PASSWORD"
  eap=PEAP
  phase1="peaplabel=0"
  phase2="auth=MSCHAPV2"
}

Modificando la voce network nel tuo wpa supplicant. 

Successivamente 

Dopo 3 mesi devi modificare lo stesso file situato in:

/data/etc/wpa_supplicant.conf

robertofabrizi, 2018-09-05:
Top @tigernero79 , that's all the info that I needed, marking this as resolved, thank you so much!

closed 2018-09-05 by robertofabrizi