motioneye-project/motioneyeos #1647: wpa_supplicant failed

lucky2004alex, 2018-09-06:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

lucky2004alex, 2018-09-10:
Already done this...10 times

Nypan-sr, 2018-09-10:
Mine looks like this in Notepad (edited out som personal info with *)
[image]
And make sure its correct before first boot.

jasaw, 2018-09-11:
@lucky2004alex Can you please check whether motionEyeOS has the TL-WN823N driver installed? Try running `dmesg` command. Also, please attach other log files here.