motioneye-project/motioneyeos #1649: Raspberry Pi Zero W starting wpa_supplicant failed (brcmfmac)

ceaswaran, 2018-09-10:
Trying to install motioneyeos on my raspberry pi Zero W
I get the following error when I boot first time

**starting wpa_supplicant failed (brcmfmac)**

My wifi Configuration on wpa_supplicant.conf file is as follows

```
country=IN
        update_config=1
        ctrl_interface=/var/run/wpa_supplicant
        
        network={
            scan_ssid=1
            ssid="MyWiFiSSID"
            psk="MyWFiPassword"
        }
```
Security Mode on my Home Wifi router is WPA-PSK/WPA2-PSK
WPA Encryption on my Home Wifi router is TKIP+AES

Next, I tried the following:
I also found my mobile hotspot with no authentication works but same mobile hotspot with WPA2 PSK doesn't work.

My wifi Configuration on wpa_supplicant.conf file **with mobile hotspot with no authentication** is as follows:
```
country=IN
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1

network={
    ssid="Mi"
    scan_ssid=1
    key_mgmt=NONE
}
```
none of the internet solution woks am again in level 1 
i tried the solution given here by you but nothing works: https://github.com/ccrisan/motioneyeos/wiki/Wifi-Preconfiguration

Please help

jasaw, 2018-09-10:
Try fixing your `wpa_supplicant.conf` indentation?
Read https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#wireless-connection-problems

ceaswaran, 2018-09-10:
Thanks
Whats the indentation mistake. I see everything to be correct
please help

[image]

jasaw, 2018-09-10:
The indentation looks correct on your screenshot. Looks like it should work.
The `wpa_supplicant.conf` file is only read on first boot. Did you try reflashing your micro SD card and writing the `wpa_supplicant.conf` file again?
Make sure your wifi access point has 2.4GHz enabled.

ceaswaran, 2018-09-11:
yes @jasaw I observed it after posting that snapshot.
Even after that change of update_config=1 t o next line . i reflashed & copied the wpa_supplicant.conf.
The result is same :( it says "wpa_supplicant failed (brcmfmac)"

andyb918, 2018-09-25:
My wpa_supplicant.conf file is exactly like ceaswaran's.  It doesn't seem to use it though.  I get the message "Connecting to wired network..." which fails, obv.  How do I get motioneyesos to ignore the wired network?

ceaswaran, 2018-09-25:
> My wpa_supplicant.conf file is exactly like ceaswaran's. It doesn't seem to use it though. I get the message "Connecting to wired network..." which fails, obv. How do I get motioneyesos to ignore the wired network?

@andyb918  Check the folllowing values 
ssid="XXXX_XXXXX"
    proto=RSN
    key_mgmt=WPA-PSK
    pairwise=CCMP TKIP
    group=CCMP TKIP
    auth_alg=OPEN
    psk="XXXXXXXXXX"
use only that suits ur need.

I am not an expert, but what I understand is above values should suit your region/wifi/internet connection standards.
**like:
proto could be either RSN (WPA2) or WPA (WPA1).
key_mgmt could be either WPA-PSK (most probably) or WPA-EAP (enterprise networks)
pairwise could be either CCMP (WPA2) or TKIP (WPA1)
auth_alg is most probably OPEN, other options are LEAP and SHARED**

More details here which was helpful for me : https://weworkweplay.com/play/automatically-connect-a-raspberry-pi-to-a-wifi-network/

TwoCanes, 2018-10-04:
I have a Raspberry Zero W.  Using this command to write 16GB uSD:

motioneyeos_writeimage.sh -d /dev/sdf -i "motioneyeos-raspberrypi-20180627.img" -n 'myssidxxx:'

I have tried everything listed on this page, nothing works.  I repeatedly get:
    wpa_supplicant failed (brcmfmac)

I've used multiple wpa_supplicant.conf files, usually like this:
    country=US
        update_config=1
        ctrl_interface=/var/run/wpa_supplicant
        network={
            scan_ssid=1
            ssid="myssidxxx"
            psk=""
        }
My wireless router works without a password (I have no neighbors.) 

I've also tried this:
update_config=1
ctrl_interface=/var/run/wpa_supplicant
network={
	scan_ssid=1
	ssid="myssidxxx"
	proto=RSN
	key_mgmt=WPA-PSK
	pairwise=CCMP
	auth_alg=OPEN
}

also with no luck.  I've tried an RPI 3 B+ (motioneyeos-raspberrypi3-20180627.img) with same results. Both boards work flawlessly in grabbing a DHCP wireless address running Raspbian Stretch. 

I've been trying to get motioneyos to work for several days and I've gone completely bald from tearing my hair out.

ceaswaran, 2018-10-05:
Yes, I too see no issues in grabbing wifi credentials running Raspbian
Also, I found this trick works for me. Create a mobile hotspot with same SSID & password as that of your home wifi.it connects first time readily with mobile hotspot, later with home wifi whenever u restart. (not sure how it works)

But we need a permanent fix from developers. This needs to be fixed.

TwoCanes, 2018-10-06:
SUCCESS!
I wrote my RPI 3 B+ with:

>     writeimage.sh -d /dev/sdf -i "motioneyeos-raspberrypi3-20180627.img" -n 'myssidxxx:myssidpwxxx'

and edited the wpa_supplicant.conf file to:

```
country=US
update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
    ssid="myssidxxx"
    psk="myssidpwxxx"
    proto=RSN
    key_mgmt=WPA-PSK
    pairwise=CCMP TKIP
    group=CCMP TKIP
    auth_alg=OPEN
}
```
and then upon boot, got wpa_supplicant: done (brcmfmac)

THEN I wrote my RPI Zero W with:

>     writeimage.sh -d /dev/sdf -i "motioneyeos-raspberrypi-20180627.img" -n 'myssidxxx:myssidpwxxx'

and edited the wpa_supplicant.conf to be exactly as above and the board booted motioneyeos.

Took about 2 weeks, :-( but I finally got it to work. ;-)

Now I don't know for sure if it was the wpa_supplicant.conf file as above or because I swapped out my old
TPLink Router for a brand-new Netgear Nighthawk AC1200 router.  And really, which has the cooler name?

I had no problems connecting to the TPLink from my RPI 3 B+, so I suspect it was using the wpa_supplicant.conf file as listed above by ceaswaran. Thanks man!

M4FFS, 2018-10-17:
Apparently it only tried to setup WiFi on the first boot, that's what worked for me

shaneporter, 2019-09-21:
Thank you @TwoCanes - I used your `wpa_supplicant` text (and changed the ssid and psk) and all was good (after failing a few times with other attempts.

TwoCanes, 2019-09-23:
On Sat, 21 Sep 2019 11:04:15 -0700
Shane Porter <notifications@github.com> wrote:

> Thank you @TwoCanes - I used your `wpa_supplicant` text (and changed
> the ssid and psk) and all was good (after failing a few times with
> other attempts.
> 

You are most welcome.  Been so long since I needed it, I forgot all
about it. ;-)

-Kevin Gross
 TwoCanes LLC
 Nevada City, CA

GitItInTheHub, 2019-11-22:
@TwoCanes (and future readers), the version that worked didn't have "scan_ssid=1" while your original one that didn't, did.

You don't want "scan_ssid=1" in your file if your wifi access point doesn't broadcast it's ssid, i.e., is hidden. Since you stated your AP is open, this fits as the reason.

I had the "wpa_supplicant failed (brcmfmac)" problem but it was because my wpa_supplicant.conf didn't have "scan_ssid=1" while my network is hidden. So when I added it then it could get on the network.

danschimpf, 2020-12-15:
I tried all of the solutions mentioned above on my mac, with no success. 

boot up would see the file, then indicate "failed (brcfmac)" error. (I was changing the extension to .conf, otherwise during boot up would not indicate "Starting wpa_supplicant:   " and would try to boot a wired connection) 

Then I tried downloading, editing and saving the file on my PC (just my specific info added to the template no additional verbiage) ..and it worked 1st time.

perhaps something with the formate of a mac txt file was unreadable.  

on the PC, I saved the file as an ANSI file

charles-cai, 2021-01-01:
guys, the easiest way to test out is to use your mobile phone's WiFi hotspot function,
1. create wpa_supplicant.conf on your C: first.  Make sure there's no hidden ".txt".  By using Notepad++, from menu "Edit - EOL Conversion - Linux (LF)", also click the menu bar "show all characters", make sure the line ending is "LF" only, NO "CR|LF".

country=US
update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
    scan_ssid=1
    ssid="your phone hotspot device name"
    psk="password of your your phone hotspot"
}

2. copy wpa_supplicant.conf to your SD card "Boot" partition, which is FAT32 and can be accessed by Windows, IMMEDIATELY after you have burn the MotionEyeOS image! My case only worked the first time you have burned the SD card.  Whenever you want to test your wpa_supplicant.conf, you have to rewrite the SD image then copy over wpa_supplicant.conf. 

This has been worked for me... otherwise even Wi-Fi worked, motioneye service still failed.

Hope this helps.

closed 2021-01-01 by starbasessd

Atashnezhad, 2021-04-16:
none of the above works! fail fail fail.

starbasessd, 2021-04-16:
Instead of highjacking a closed issue, how about opening a new issue, answer all the default questions, (even reference this issue if needed). For PiZero please also include which version of the PiZero found on the back of the Pi.