motioneye-project/motioneyeos #1651: static ip on Access Point mode

linoreale, 2018-09-12:
Raspberry PI 3 B+
Hello,
first of all I would thanks the author of Motioneyeos for his great job.
I'm trying to set my RPI into an Access Point without dhcp support and I would know if there's a way to specify a static ip address into hostapd.conf or dnsmasq.conf.
If not, how can I set a static ip to the AP otherwise?
Thank you and regards.

ccrisan, 2018-09-12:
You are right: when in "networkless" mode, if you want to start an AP you have no (easy) way to set up the IP address of the AP itself. I think best way to implement this would be by setting it up in 
[static_ip.conf](https://github.com/ccrisan/motioneyeos/wiki/static_ip.conf) and reading it from the dnsmasq or hostapd init script.

linoreale, 2018-09-12:
Hi Calin,
thanks for your reply.
For "networkless" I think you mean with no LAN  connections, don't you?
Trying to assign an ip to the wlan0 I've already tried putting the static_ip.conf on boot partition with hostapd.conf and dnsmasq.conf, but no "os_networkless", and on boot Raspberry always loops on reboots.
How can I manage with it?
Regards

ccrisan, 2018-09-12:
As I said, you can't do that out-of-the-box right now. You could try to manually configure your wlan0 interface in [userinit.sh](https://github.com/ccrisan/motioneyeos/wiki/Tweaks#adding-custom-startup-commands).

Something like `ifconfig wlan0 192.168.0.1 up` should do it. Keep us posted.