motioneye-project/motioneyeos #1652: "Accessing network share failed: cannot create root directory" but settings are correct

robertofabrizi, 2018-09-12:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20180627

### Board Model

I am using the following board/model:Raspberry PI ZERO W

The camera feed is working, but I can't save it to my Nas share even though the config looks just fine to me.

This network share works in Windows and is already created: \\\192.168.1.3\Recordings\Camera1
This one isn't yet created: \\192.168.1.3\Recordings\Camera2

How do I have to insert either one of those it in the UI?
Like this 

[image]

doesn't cause a "Accessing network share failed: cannot mount network share!", but still doesn't work as it says "Accessing network share failed: cannot create root directory"

Thank you and have a nice evening,
Roberto

HOD42, 2019-09-08:
I have a similar issue. I've added a specific "cams" user to my (Raspberry Pi) NAS and to smb itself. Although the "cams" user can access the file share if I leave the root directory as "/", if I change this to "/cams" (or other) it fails to create the root directory. The same thing happens if I use my root user credentials. I'm using SMB3.0 BTW.

HOD42, 2019-09-15:
I found what my issue was. I needed to mount the filesystem on each of my MotionEye RasPis to r/w in order to add the SSH key (to known hosts). Once I'd done a test ssh cams@NAS and saved the fingerprint it was fine. I even remembered to remount the filesystem read-only afterwards!

ir-fuel, 2019-10-12:
How did you solve this? I get a "read only filesystem" error if I want to do this.

ir-fuel, 2019-10-12:
How do you switch the filesystem to read-write?

Essaysee, 2019-10-12:
Google it. Do some looking. Someone has already asked this question and the
developer himself has explained how to mount the partition to read-write.

justinbayliss, 2019-12-03:
> Google it. Do some looking. Someone has already asked this question and the developer himself has explained how to mount the partition to read-write.
> […](#)

People need to stop with the whole "Google It" response. If you have nothing to contribute, DON'T. The issue is that when I do google it, this page comes up towards the top. The exact issue I am having. Then it just dead ends with someone saying "Google It" Well guess what" I DID!" and around and around we go. If nothing else provide a link to the page that actually solves or addresses the issue. Thanks for nothing, but getting me upset that you helped in no way!

tob2bottom, 2020-02-16:
Hi guys,

I had the same issue.
Then I tried to add the subfolder to the **Share Name** path. Now it's working.
[image]

Maybe this will help someone.

Best regards
top2bottom

snowflakeas, 2020-03-01:
> Hi guys,
> 
> I had the same issue.
> Then I tried to add the subfolder to the **Share Name** path. Now it's working.
> [image]
> 
> Maybe this will help someone.
> 
> Best regards
> top2bottom

Tnx - It helped me - hope that this can be fixed, as it seems to be a trivial bug
UI wise: "share name" should just be the name of the share
"Root Directory" should be named "Motioneye Directory" or similar and should allow for the Camera_2 subdir as per solution above. Also nice if the SW would autoappend / remove '/' if necessary - never clear if needed.

such that "root directory" can just be named "directory" and work as expected, "Share Name" should not include the

astewardson, 2020-03-30:
Thank you top2bottom, been having this issue as well and this worked perfectly!!

JulianMHall, 2020-04-11:
This worked for me too. In short:
root directory: /
Share name: [format is Share Name/path/to/file] In my case this was DEMETER/Security/Cam1 as my share is DEMETER and the folder on that share is Security with sub folders for Cam2 and Cam3.
I have not set up a surveillance username yet but that's a good idea.
Kind regards,
Julian

Ryadovoy001, 2020-06-04:
This worked for me too.
But now stumbling on a different issue.
I've set the recording mode to continuous recording for 900 seconds (15 minutes) and it stops after storing one file to the SMB share (SMB3.0 set in motioneye).
On server side I can see that the motioneye user is still connected through CIFS (don't know why).
Btw NAS is a Synology.
[image]
[image]
[image]

JulianMHall, 2020-06-04:
Hi Ryadovoy001,
The first thing that comes to mind is file size, in Mb not minutes of course as the latter could vary with compression. Do they all quit recording at the same file size? That could be a SMB limitation. I use a Synology too, but my shares are NFS not SMB. I don't have screen shots at present but that's really the only difference between our installations.
Kind regards,
Julian

Ryadovoy001, 2020-06-04:
Hello JulianMHall,
The recordings stop at varying file sizes. Anyway was hoping this easy way of setting up a Network Share.
Was hoping to set it up the easy way as I was just playing around with leftover rpi3 and motioneye os.
Anyway will have to set up NFS for this, but that will take some small research on my side on how to implement this on motioneye os.
Kind regards

JulianMHall, 2020-06-04:
NFS is transparent on Motioneye, the screenshots further up this thread (and my earlier post on 11 April) give the settings. The harder (and still pretty easy) part is the Synology sharing NFS, but it defaults to that anyway as the NAS runs on Linux.

snipes040, 2021-03-03:
> Hi guys,
> 
> I had the same issue.
> Then I tried to add the subfolder to the **Share Name** path. Now it's working.
> [image]
> 
> Maybe this will help someone.
> 
> Best regards
> top2bottom

This fixed it for me too!  Seems like a bug but this definitely works thanks!

closed 2021-03-03 by starbasessd