motioneye-project/motioneyeos #1657: One video files from two cameras together?

andreluisos, 2018-09-19:
I have a huge gate which can't be covered with one camera. Is it possible to merge two câmeras into one video output?

Another question: is it possible to tweak brightness, contrast and iso through commands? If yes, how do I do it?

ccrisan, 2018-09-19:
> I have a huge gate which can't be covered with one camera. Is it possible to merge two câmeras into one video output?

 Not to my knowledge.

> Another question: is it possible to tweak brightness, contrast and iso through commands? If yes, how do I do it?

See the `v4l2-ctl` command.

closed 2018-09-19 by ccrisan

andreluisos, 2018-09-19:
Sorry. I tried "v4l2-ctl -d /dev/video0 -c brightness=60", for instance, but it didn't change camera brightness.