motioneye-project/motioneyeos #1658: how to configure an RTC DS3132

linoreale, 2018-09-21:
Hi,
I am using Motioneyeos on a Raspberry PI 3 B+ as a standalone NVR, without internet connection.
So I bought an RTC DS3132 and now I  would configure it on the board.
Could you please help me to do this?
Thanks

ccrisan, 2018-09-21:
Have you tried [this](https://github.com/ccrisan/motioneyeos/wiki/Tweaks#adding-a-rtc-module)?

linoreale, 2018-09-21:
Hi Calin,
no, not yet.
I thought ds1307 was different from ds3231.
Is it the same?
Thanks a lot and regards

ccrisan, 2018-09-21:
@linoreale no idea, but they are probably not the same. However I believe the setup should be similar.

linoreale, 2018-09-21:
Has anyone ever tried using a ds 3231 on motioneyeos?
Thanks

linoreale, 2018-11-14:
Hi,
I've just installed a DS1307 rtc module and followed [your instructions](https://github.com/ccrisan/motioneyeos/wiki/Tweaks#adding-a-rtc-module) to configure it but, when I reboot the RPI, it still looks for ah http server to set system time.
Is there a way to avoid this?
Could you tell me how please?
Thanks and Regards

jasaw, 2018-11-14:
@linoreale Enabling the RTC module does not change how system time is synced. All that the RTC does is just to give the correct system time at boot up, before time is synced.

This is how it works:
- you save the correct time on the RTC which is battery backed. `hwlock -uw` will write the system time in UTC to RTC, so make sure your system time is correct before running the command. Use `hwlock -r` to read the RTC time to verify that time is set correctly. Not sure if you need the `-u` option here.
- when the device boots, it uses the RTC time as the system time by running `hwlock -s`.
- you are free to sync time via any protocol you want, http or ntp, or not sync time at all.
- RTC time will drift slowly, which means loading a more accurate time onto the RTC once in a while. How much the RTC drifts depends entirely on how good your RTC is.

Back to your question, I'm not sure if there's a way to disable time sync from MotionEye web front-end. @ccrisan might be able to point you to the right direction.

linoreale, 2018-11-14:
Hi jasaw,
Have I to use "hwlock" or maybe "hwclock"?
Thank you so much for your reply
Regards

jasaw, 2018-11-14:
oops, should be hwclock. Sorry.

ccrisan, 2018-11-16:
@linoreale unfortunately there's no (easy) way to disable date/time syncing from the UI. If you really do not wish your system to be updated from the internet, you can always set the *Expert Settings > Date HTTP Host* to something invalid. It will add errors to your log files but won't restart your system.

linoreale, 2018-11-30:
Grazie!

alexxtasi, 2019-06-20:
Hi all
I would like to use MotionEyeOS in a similar system setup... having a Raspberry Pi Zero completely seperate with no internet connection.

Would the lack of HTTP / NTP date method, or even the non-existence of an RTC module cause problems booting and operating the system ? (apart from the wrong time I see on the screen !)

thanks in advance

ashseven, 2019-10-21:
I "tried" many RTC modules included DS3231 from Seed Studio. Well, reading the instructions, for a proper work, you have to install driver for it. However, in according with the github page of motionEyeOS, not possible to install:

> Can I install <some package>?
> 
> No, you cannot. motionEyeOS is not a general purpose Linux distribution, but rather a camera firmware. Moreover, its system partition is read-only and should not be modified.
> 
> Your best option is to use motionEye on top of a general purpose Linux distro, on which you'll probably be able to install your favorite packages.
> 
> If you really want your package to be part of motionEyeOS and you have the required skills, you could build motionEyeOS from source, after adding your package to the BuildRoot configuration. 

I thought to escape from this limitation, making my own version of motionEyeOS, but what i read in the messages before mine, scared me. Especially when people speaking about how the time/date are sync.
I have three different RTC modules; DS1307, DS3231 and again a [DS1307](https://wiki.52pi.com/index.php/DS1307_RTC_Module_with_BAT_for_Raspberry_Pi_SKU:_EP-0059)
well, if someone have ideas and want that i try as a tester, i will do with pleasure.

ThomasManthey, 2019-10-23:
I have a very similar problem because I want to use motionEye as a autonomous camera that connects automatically to my mobile hot spot to access the collected data. Please see the ticket "Running MotioneyOs dose not came up without internet connection #2108"

BirdiPiduino, 2022-09-12:
> Hi, I've just installed a DS1307 rtc module and followed [your instructions](https://github.com/ccrisan/motioneyeos/wiki/Tweaks#adding-a-rtc-module) to configure it but, when I reboot the RPI, it still looks for ah http server to set system time and the rtc module doesn't work. Is there a way to avoid this? Could you tell me how please? Thanks and Regards

I used the sames instructions with a Raspberry Pi Zero and a RTC DS1307. **It work.**
note : to avoid the interruptions of the system when internet is not available, my RPi is configured in [Access Point (AP)](https://groups.google.com/g/motioneye/c/cM9PLChEVyo).

After [those same instructions](https://github.com/motioneye-project/motioneyeos/wiki/Tweaks#adding-a-rtc-module), I was connected to my RPi with Putty and I have set manually the date and time with those instructions
`date -s "2013-11-19 15:11:40"`
`hwclock -w`
 
good luck