motioneye-project/motioneyeos #1659: Camera Pi stores files while configured network-storage

Rycoviac, 2018-09-21:
I've configured my cam to store files on network storage. The Hub is a PI3 with motionEyeOS installed.
Storing files on the hub works, but I've found out, that the cam is saving also files on /data/media. It seems, it transfers the mediafiles after a completed video to the Hub, but don't delete the files after that. The videofiles still stored on cam-sdcard. Since it is a 32GB card on cam, theres plenty of storage. But it could be a problem on longterm, since I've to maintain it regulary.
Did I miss something?

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.
Confirm both.

### motionEyeOS Version

I am running motionEyeOS version: dev20180730.

### Board Model

I am using the following board/model: Raspberry PI Zero WH for Cam, PI 3 for Hub.

### Camera

I am using the following type of camera: MMAL.

My camera model is: Electreeks Raspberry Pi Kamera Modul Full HD mit Infrarot LEDs.

### Network Connection

My motionEyeOS unit is connected to the network via: PI Zero on WLAN, Pi3 on LAN

### Peripherals

none

### Log Files

ccrisan, 2018-09-21:
> I've configured my cam to store files on network storage. The Hub is a PI3 with motionEyeOS installed.
Storing files on the hub works, but I've found out, that the cam is saving also files on /data/media.

That is very unlikely, if I understand your setup correctly. Can we see some configuration screenshots from your hub, just so that we're clear about where your media files should go?

Rycoviac, 2018-09-27:
Hm, before I post giant screenshots... could it be, that I have to set at "Movies" the preserve to "one day", so after 24h the files will be deleted local on the cam (but not on the hub)? At the moment I'm at prototyping, so I diddn't test this jet.

Rycoviac, 2018-09-27:
[image]
[image]
[image]

ccrisan, 2018-09-29:
I'm not saying that this is the cause of your problem, but the Network Share name should definitely not be made of more than one path component. In your case, that would probably be `storage`. The `cam1` part should probably go on the Root Directory. Try it that way and see how it behaves.

Rycoviac, 2018-10-02:
Ohh... I realised that all under /storage on the Cam-Pi is a link to the HUB. This isn't a own directory on the Cam1-SDcard, it is the directory on the Hub. I don't know exactly why you see this on FTP while on the cam... thats new for me, but something cool.
So there are no files on Cam1-SD. Case closed =D

closed 2018-10-02 by ccrisan