motioneye-project/motioneyeos #1661: Motioneyeos Webserver and MJPEG don't respond, ping still works on Raspberry PI Zero W

Heinzelschelm, 2018-09-22:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

ccrisan, 2018-09-29:
@Heinzelschelm you're not the only one to report responsiveness issues with motionEye lately. Unfortunately I simply have no idea what could cause them and what would be the best way to identify this problem.

One thing that I keep seeing in users' logs when they experience this issue is timeout when communicating with Motion:

    2018-09-21 17:48:56: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081

If you look closely in your motioneye.log, you'll see that whenever you have a `hello!` that is **not** preceded by a `bye!` (i.e. not a clean reboot), you'll always have a similar Motion timeout instead. This leads me to believe that Motion may be the one misbehaving.

One thing you could do is to enable debugging (in Expert Settings) and then you'll get more useful stuff in motion.log.

Dutchs, 2018-10-05:
Having very similar symptoms on a rpi3b+ seemingly locking up SSH, webserver and FTP after about a day both with latest release 20180627 and dev20180730
Last time it hang was during a motion event i suspect, due to that clip being corrupted in the progress
Here are logs, with MotionEyeOS set with debug logging turned on
[boot.log](https://github.com/ccrisan/motioneyeos/files/2448584/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/2448585/dmesg.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/2448586/messages.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/2448587/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/2448588/motioneye.log)

boardtim01, 2018-10-16:
Where are all these log files located? I only see the /var/log/motion.log.

Rubberduck84, 2018-11-05:
I have the same problem with the raspberry pi 3b + i uploaded the images to google drive.
This are my logs.

[motioneye (4).log](https://github.com/ccrisan/motioneyeos/files/2547621/motioneye.4.log)
[boot (6).log](https://github.com/ccrisan/motioneyeos/files/2547628/boot.6.log)
[motion (4).log](https://github.com/ccrisan/motioneyeos/files/2547623/motion.4.log)
[messages (4).log](https://github.com/ccrisan/motioneyeos/files/2547624/messages.4.log)
[dmesg (4).log](https://github.com/ccrisan/motioneyeos/files/2547625/dmesg.4.log)

Ik hope that anybody fix this problem.

Rubberduck84, 2018-11-05:
> Where are all these log files located? I only see the /var/log/motion.log.

under the tap expert settings!

ccrisan, 2018-11-05:
@Rubberduck84 I see " user.notice wifi: disconnected" in your log - check your wifi connectivity.

Rubberduck84, 2018-11-05:
Thanks Ccrisan,

I will try it with cable.
Thanks for your quick response.

Rubberduck84, 2018-11-28:
> Same here.
> I also get this error message in motioneye.log:
> `2018-11-27 17:48:56: [motioneye] ERROR: mjpg client timed out receiving data for camera 1 on port 8081`
> 
> A Few minutes later I got:
> `2018-11-27 17:50:04: [motioneye] ERROR: motion not running, starting it`
> 
> After that the message
> 
> ```
> ERROR: mjpg client timed out receiving data for camera 1 on port 8081
> ```
> is repeating every 10 seconds.
> 
> If I reboot the cam with webinterface, the cam doesn't come up.
> I have to power down and up, than the cam is working again

Try it with a utp cable it works for me.

helmar74, 2018-11-28:
A other cam, which is also connected with wifi doesn't have this issue.

Rubberduck84, 2018-11-28:
> A other cam, which is also connected with wifi doesn't have this issue.

Have the wifi connection the same distance?

helmar74, 2018-11-28:
yes, both cams are located in the sam room (garage) only 2 meters distance is between them

Rubberduck84, 2018-11-28:
I mean how far its the distance to your wifi router?
And do you use motion detection?

helmar74, 2018-11-28:
The related access point is also in the garage. The cams are connected to the AP

Rubberduck84, 2018-11-28:
Please flash the software on a new sd card  and try it agian.
i use this type of card. Samsung EVO Plus Micro SDXC 64GB bis zu 100MB/s Class 10 U3

TexanScot, 2018-12-22:
Has anyone come up with a solution for this issue.  Motioneye running on my Pi Zero W seems to 'lock up' (can't find server and can't SSH to the Pi) after running perfectly for a while.  I have to reboot the Pi in order to get Motioneye working again.

What seems strange is that system seems still to be working - when I reboot I can see pictures captured when the Pi was apparently unresponsive. I need to check further that this truly is the case however.

jasaw, 2019-01-03:
@TexanScot Please check your wifi connectivity just to rule it out. Make sure your wifi AP is stable, and not too far from your Pi Zero W. The Pi Zero W wifi antenna is pretty bad.

marhau, 2019-04-13:
Same behaviour here with a Pi Zero W and motionEye Version 0.39.2, Motion Version 4.1.1 and OS Version motionEyeOS 20180627. Pinging the Pi worked, but everything else timed out. Just got my hands on the powersupply, so its online until next freeze..

To me, it looks like the system hangs after getting "crng init done" in dmesg.log, but the timestamps aren't up to date, so I'm just guessing.

dmesg.log
`[Thu Jan  1 01:05:35 1970] random: crng init done
---- booting motionEyeOS 20180627 ----`

motioneye.log:
`2019-03-31 03:13:48: [motioneye]    DEBUG: 200 HEAD / (127.0.0.1) 10.82ms
2019-03-31 03:13:53: [motioneye]    DEBUG: 200 HEAD / (127.0.0.1) 11.53ms
2019-04-13 20:48:59: [motioneye]     INFO: hello! this is motionEye server 0.39.2
2019-04-13 20:49:00: [motioneye]    DEBUG: using motion version 4.1.1
2019-04-13 20:49:01: [motioneye]    DEBUG: using ffmpeg version 3.3.5`

The other logs doesn't reveal anything as far as I can see. Except for messages.log which is unavaiable for some reason.

Maybe the solution is a relay cutting the power once a day.