motioneye-project/motioneyeos #1664: daemon.err dhclient: send_packet: No buffer space available /

norey, 2018-09-27:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20180627

### Board Model

I am using the following board/model: Raspberry Pi 0 w

### Camera

I am using the following type of camera: Fast Network Camera and Simple MJPEG Camera

My camera model is: PI NOIR CAMERA V2

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

None

### Log Files

I consider the following log files relevant to this issue:

 * dmesg: https://bpaste.net/show/2bee2f83eac7
 * /var/log/messages - https://bpaste.net/show/69c4fd688f98
 
When first powered on, things are fine. After a while, the pi is not reachable anymore until I reboot it. I checked the logs and found the above.

ccrisan, 2018-09-29:
@norey links to your log files are (now) broken. Please attach them here directly or set an unlimited expiration time.

norey, 2018-09-29:
I cannot extract older dmesg info. 
This is what I found in /var/log/messages - https://bpaste.net/show/a9812f6384d8

ccrisan, 2018-09-29:
@norey the error you're getting is not something that we see every day on a Linux machine. A [quick googling](https://askubuntu.com/questions/210451/what-does-ping-sendmsg-no-buffer-space-available-mean) indicates that this may be a hardware fault. You could try to tweak `/proc/sys/net/core/wmem_max` but it's very unlikely to actually fix the issue here.

norey, 2018-09-30:
That's unfortunate, it is a new rpi0w that hasn't been used much.