motioneye-project/motioneyeos #1667: Mass storage not recognised

Burgy22, 2018-10-03:
Preliminary Docs

I confirm that I have read the CONTRIBUTING guide before opening this issue.

I confirm that I have read the FAQ before opening this issue.
motionEyeOS Version

I am running motionEyeOS version:20180627
Board Model pi3

I am using the following board/model:Raspberry pi 0w connected with fast networking to Raspberry pi 3B+ the mass storage hub
Camera

My camera model is:Raspberry pi camera v1.3
Network Connection

My motionEyeOS unit is connected to the network via:WiFi 

I am using the following peripherals that I consider relevant to this issue:
1Tb External Hard Disk (planed for mass storage)
Tried using the following formats, without success: fat32, ntfs, ext4.
Mass storage does not show up in the storage device pulldown list?
Log Files

I consider no log files relevant to this issue:

ccrisan, 2018-10-03:
@Burgy22 log files please.

Burgy22, 2018-10-04:
During installing (first boot) no Folders were produced on the Hard Disk, now done manually.
The Mass Storage is now in the pulldown list an can be used.
@ccrisan 
Many thanks for your excelent work..

closed 2018-10-04 by Burgy22