motioneye-project/motioneyeos #1669: IP Camera stream in low quality on motionEyeOS

wgergeus, 2018-10-04:
I am running motionEyeOS version: (dev20180730). on Pi3 and i tried to connect an IP Camera to motionEyeOS .. 
the IP Camera working but only stream in low quality (**bad stream quality**)

_i attatched 3 photos for the cam and its sittings_
[image]
[image]
[image]

**_so how to get it working with good quality ?_**

**NOTE**:
the IP Camera working on its phone app called (yoosoo) with good quality
and this photo from my phone
[image]

jasaw, 2018-10-05:
IP cameras typically have 2 streams, a low quality for motion detection, and a high quality for recording. I'm pretty sure motion software supports this, but I have never set one up before. Maybe someone else might be able to help you with the config.

alexei-developer, 2018-10-31:
Also interesting, I am not find where set stream for motion detection and where set stream for write video. Please, help. =) It is possible?

Chiny91, 2018-10-31:
Re @jasaw comment... that would be useful... peruse of the motion words... oh yes:
https://motion-project.github.io/motion_config.html#netcam_highres
I'd never noticed that before, thanks.

I need my system to run without fail in the next few weeks, but should be able to test by Christmas.   Of course by then, someone else will already have reported back 😄

alexei-developer, 2018-11-01:
Thanks, but.... if i am add extra option
netcam_highres rtsp://192.168.0.50:554/user=admin&password=&channel=0&stream=0.sdp
or 
netcam_highres "rtsp://192.168.0.50:554/user=admin&password=&channel=0&stream=0.sdp"
myeyos write cannot open device... =(

[image]
otherwise if not extra option, then open device success:
[image]

Camera device set is stream 1: CIF (rtsp://192.168.0.50:554/user=admin&password=&channel=0&stream=1.sdp)
For extra option set steram 0: 1080

alexei-developer, 2018-11-01:
All works. It is my camera very long connect in 2Mp and VBR mode... =(

Chiny91, 2018-11-01:
Glad it worked.

I am testing netcam_highres myself and so far, so good.   To make it work:
(1) Use the low res URL as the Video Device in motionEyeOS
(2) Add an Extra Motion Option of `netcam_highres your_high_res_URL`
(3) Ensure the cam config delivers the correct video stream.

Some big files can be produced, which may be a problem for some.

wgergeus, 2018-11-05:
i used this RTSP url "rtsp://192.168.8.109:554/onvif2" and it was the low quality image
and replaced to "rtsp://192.168.8.109:554/onvif1" with Video Resolution (800×600) the steam quality was good but when i move i get bad image 
[image]

**how can i fix this Issue?**

ccrisan, 2018-11-05:
@wgergeus streaming issues when using RTSP are often reported when using UDP instead of TCP. Please try adding the camera using the TCP protocol (if you did not do that already). If the problem still persists, please [report a Motion issue](https://github.com/ccrisan/motioneyeos/wiki/Report-Motion-Issue).

wgergeus, 2018-11-07:
@ccrisan thanks for your comment.
i tried to use the TCP protocol  but it doesn't work 
**unable to open video device**

in only work on UDP protocol  on this url : 

rtsp://192.168.8.109:554/onvif1

what can i do to solve this problem?

and i have another issue:

when tried to use motion notification by E-Mail i want to activate the notification at night only so if i tried to open TLS option the notification works but when i turn it off the system can't connect to wifi network and to solve that issue i had to format the SD Card and install motioneye os again 

so what can i do?

ccrisan, 2018-11-07:
@wgergeus as I said in my previous comment, please [report a Motion issue](https://github.com/ccrisan/motioneyeos/wiki/Report-Motion-Issue).

alexei-developer, 2018-11-08:
How change current add cameras from UDP to TCP? I am not find it in settings:
[image]

ccrisan, 2018-11-08:
You need to remove and readd the camera.

closed 2018-11-08 by ccrisan

alexei-developer, 2018-11-08:
Oh, and add all cameras again with set all settings again? =(((

Pilou72, 2019-04-29:
@wgergeus : Hi, I have exactly the same problem, with a Yoosee IP Camera freshly bought (Good quality with the Yoosee CMS, but very bad when movements with MotionEyeOS). Did you succeed in solving this problem ? Did you already report a Motion Issue ?

Thks,
Pierre