motioneye-project/motioneyeos #1671: motionEye: CANT ACCESS to added in-browser video playback

fetienne, 2018-10-08:
### motionEyeOS Version
I am running motionEyeOS version: (motionEyeOS 20180627).

### Board Model
I am using the following board/model: (raspberry 0).

### Camera

### Network Connection
My motionEyeOS unit is connected to the network via: (WiFi).

### Peripherals

### Log Files

Where I could find the added in-browser video playback ?
I can't see it under firefox.

Thank's

ccrisan, 2018-10-12:
@fetienne in the media browser, as soon as you click on a recorded movie. If you don't see a play icon, then your browser probably does not support HTML5 video playback.

closed 2018-10-12 by ccrisan

fetienne, 2018-10-12:
Hi,

But I try on Firefox 62.0.3, Opera 56.0.3051.40 and Chrome 69.0.3497 on windows and Samsung internet 7.4.00.70 under Android.
On both, I can't see a play icon. Only Close and Download.

Which browser is compatible for that ?

Thank's.

ccrisan, 2018-10-13:
Screenshot please.

EOUpSL93Y, 2019-09-04:
Same problem on latest Motioneye. https://pic4a.ru/99/xyV.png