motioneye-project/motioneyeos #1676: Problems booting up

davekpython, 2018-10-14:
Hello,

My Motioneyeos boots up to the login but I cannot see it on my laptop. I have the IP address and nothing shows up.
There seems to be  a lot of failures during the boot up process. i've attached a screen shot of the last lines of code at boot up.

[image]

I've formatted and reformatted the sd card and downloaded the latest version many times.

Hope you can help.

Thank you.

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version
motioneyeos-raspberrypi2-20180627.img.xz

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model
I am using the following board/model: Raspberry PI 2b.

### Camera
raspberry pi noIR camera v2

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection
wireless

My motionEyeOS unit is connected to the network via: wifi.

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

ccrisan, 2018-10-15:
Please try to take a picture of the beginning of the log, **at the very first boot** after writing the OS image. Your data partition is not being created and that results in all the painful errors you see.

You could "scroll" through the console log from an attached keyboard using Shift + Page Up/Down.

davekpython, 2018-10-15:
Yes, I think that's correct. It seems it's not creating a lot of directories. (here's a photo at top).

Thank you.

[image]

ccrisan, 2018-10-15:
This seems like a photo from a subsequent failing boot. I need one from the very first boot right after you have written the OS image from scratch to your card.

davekpython, 2018-10-15:
In an effort to problem solve, I tried reformatting the SD Card and found it to be corrupted.

So I just picked up a new Sandisk Extreme 32gb sd card. 

I did not format it as it was already formatted. The win32disk imager copied the motioneyeos-raspberrypi2-20180627 img to the sd card. I extracted fwupdater zip to get start.elf and rootfs.cpio.  Then put the sd card into the raspberry pi b2. I attached the raspberry pi to the router with an ethernet cord and booted up the sd card.  (see below image for screen shot.

It seems like it should be working. But, I can't "see" my rasp pi ip address. I know what it is but it doesn't show up on my angry ip scanner (it used to) and when I put the address in the browser it doesn't find it.

I did notice there is a kernal img folder as one of the files on the sd card (see image below) I wonder if this might have something to do with the problem. should i have copied the img file to my computer then used the windisc imager to extract it to the card?

Thoughts?

Thank you.

[image]

[image]

davekpython, 2018-10-15:
Update.

I'm happy to say I have it up and running (the new sd card made the difference... mostly.

I have pointed my browser to the pi's ip address and my motioneye page is right there but there is no image from the camera. see photo 
[image]

I installed the camera before working with motioneye so i know it works.

thoughts?

Thank you.
glad to be getting close.

davekpython, 2018-10-16:
update

reinstalled camera

program working fine and I have image.

Looking forward to learning about motioneye.

Thank you.

closed 2018-10-16 by davekpython