motioneye-project/motioneyeos #1680: Unable to open video device

mianrelie, 2018-10-17:
### motionEyeOS Version

I am running motionEyeOS version: **motioneyeos-raspberrypi3-20180627**

### Board Model

I am using the following board/model: **Raspberry Pi 3B+**

### Camera

I am using the following type of camera: **Network Camera**

My camera model is: **Android Mobile Phone + App "IP Webcam"**

### Network Connection

My motionEyeOS unit is connected to the network via: **Ethernet**

[image]

Issue with EyeMotionOS Version?

ccrisan, 2018-10-17:
Logs please (all of them).

mianrelie, 2018-10-17:
[motionlog.txt](https://github.com/ccrisan/motioneyeos/files/2489026/motionlog.txt)
[bootlog.txt](https://github.com/ccrisan/motioneyeos/files/2489067/bootlog.txt)

mianrelie, 2018-10-18:
It solved with motioneyeos-raspberrypi3-dev20181008 but...haven't sound :(