motioneye-project/motioneyeos #1682: A slightly different RPi Zero W wifi issue, I hope: No device found

JoeHartley3, 2018-10-17:
I am trying to run  motionEyeOS version 20180627 on my RPi Zero W v1.1

I've read all the docs and issues on this, and I believe my connectivity issue may be a bit different than others - I hope it's easily resolvable.

I used the writeimage.sh script with the -n SSID:PSK argument to get a bootable MicroSD card.  I then mount /dev/mmcblk0p0 and copy a wpa_supplicant.conf file to it, umount and give it a try.

What I get is a bootloop, but I never see the "starting wpa_supplicant failed (brcmfmac)" error others seem to get.  This is what the RPi shows me just before it reboots:

    configuring wired network: no device
    network: no network connection available, rebooting

The other thing that I thought might be of issue is an improperly formatted config file, so I actually copied the wpa_supplicant.conf file from a RPi 3B+ running Raspbian Stretch:

`ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
        ssid="MYSSID"
        psk="XXXXXXXXXX"
        key_mgmt=WPA-PSK
}
`

Has anyone seen this partucular variant before?
Thanks!

ccrisan, 2018-10-18:
@JoeHartley3 make sure you provide the wpa_supplicant.conf file *before the very first boot*.

JoeHartley3, 2018-10-18:
I had seen the discussion on how the file is picked up from /boot then removed, so I was mounting the SD card immediately after running writeimage.sh.

I ran into an issue with the SD slot on the laptop I use to write images; it may be that my later invocations of writeimage.sh failed silently at the dd command, I will be testing that this evening and will update this issue based on the results.

JoeHartley3, 2018-10-20:
I verified that my SD card writer on my laptop failed silently at some point, but came back after a reboot and now I'm up and running as written on the tin.  Thanks!

closed 2018-10-20 by JoeHartley3