motioneye-project/motioneyeos #1683: I dont habe any picture

PhoenixEX, 2018-10-18:
Hello to each other,

so first of all, I'm an absolute beginner.
I bought the PI for surveillance and really don't know much about it yet.
Therefore it would be nice if you could explain everything very simply.
I bought the PI and a corresponding camera and set up motioneyeOS as shown in many tutorials.
The problem:
I don't get any pictures displayed!!
Can anyone say why?
What do I want to do with it?
Video recordings in front of my front door, which are stored 1-2 days

### motionEyeOS Version

motioneyeos-raspberrypi3-dev20181008

### Board Model

RPI3b

### Camera

Bought from here:
https://www.amazon.de/Electreeks-Raspberry-Kamera-Modul-Infrarot/dp/B0763Q5ZBS/ref=sr_1_sc_1?s=computers&ie=UTF8&qid=1539882958&sr=1-1-spell&keywords=infaror+rpi3

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via:
Ethernet

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

SD Card 128GB

### Log Files

I consider the following log files relevant to this issue:

How can I copy my log files????

Other Question:
Which camery-typ should i add?
Local V4L2 or MMAL

ccrisan, 2018-10-19:
Let me answer all of your issues, one by one.

> Therefore it would be nice if you could explain everything very simply.

Please use our [Google group](https://groups.google.com/forum/#!forum/motioneye) as well as our [wiki](https://github.com/ccrisan/motioneyeos/wiki) for questions and docs.

> motioneyeos-raspberrypi3-dev20181008

While we appreciate users testing our nightly builds, I believe it's best that you use the latest stable release while getting started with your setup.

> Bought from here:
https://www.amazon.de/Electreeks-Raspberry-Kamera-Modul-Infrarot/dp/B0763Q5ZBS/ref=sr_1_sc_1?s=computers&ie=UTF8&qid=1539882958&sr=1-1-spell&keywords=infaror+rpi3

The camera looks fine and I think that it should work well with any RPi model. However, please test it with Raspbian before further investigating this issue.

> How can I copy my log files????

Please read the [contributing guide](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md), a link which you shouldn't have removed from the top of the issue.

> Which camery-typ should i add?

Judging from the link you gave us, probably a MMAL camera. It should however work almost as well with V4L2.

PhoenixEX, 2018-10-19:
Hey ccrisian,

first of all big thx for your help.
Ok i will try a stable Version.
Is this version stable?
20180627
[https://github.com/ccrisan/motioneyeos/releases](url)

marrobHD, 2018-10-20:
I have the same issue with the camera. The original raspicam isnt working anyway. 
I tryed raspistill - o image.jpg
I got
nmal: nmal_vc_component_enable: failed to enable component: EMOSPC
 nmal: camera component could´t be enabled
 nmal: main: Failed to create camera component
 nmal: Failed to run camera app. Please check for firmware updates

ccrisan, 2018-10-22:
Again, please check with Raspbian and report back. If it doesn't work, then it's not a motionEyeOS issue.