motioneye-project/motioneyeos #1684: Old version can't be updated.

snigel, 2018-10-20:
Running motioneye 20160801 on a RPI1 trying to do an update.
It fails with a generic; "The update process has failed!"

Looking at the log you can see that the wget download fails trying to download,
https://github.com/ccrisan/motioneyeos/releases/download/20180627/motioneyeos-raspberrypi-20180627.img.gz

That url is broken, since only a .img.xz exists. Looking at update.py one can see that the update string is hardcoded to only look for .img.gz

Is there any way I can patch the installation so I can perform an update without reinstalling?

ccrisan, 2018-10-22:
Update incrementally to intermediate versions. If that doesn't work, write the OS from scratch.

closed 2018-10-22 by ccrisan

ccrisan, 2018-10-28:
@snigel I'm really sorry that you had/still have to climb a ladder and physically intervene on your RPi unit, but things have changed considerably in two years. We simply cannot provide full backwards compatibility between all versions, no matter how old.

snigel, 2018-10-28:
@ccrisan no blame intended! Sorry if it looked that way.
Big thanks for the support and software.