motioneye-project/motioneyeos #1690: number of columns option missing from preferences

RHTizzy, 2018-10-26:
Had two colums and 1 row, but changed layout to 1 column and 2 rows to check if it was practical on an Android tablet using Chrome. But then the option for 2 columns disappeared and does not reappear. Tablet is most often used in landscape mode and it is more practical to use 2 columns. Highly peculiar that that option would need to be hidden in the GUI 

Preliminary Docs
I confirm that I have read the CONTRIBUTING guide before opening this issue.

I confirm that I have read the FAQ before opening this issue.

motionEyeOS Version
I am running motionEyeOS version: 0.39.3

Board Model
Pi Zero W

Camera
I am using the following type of camera: MMAL

My camera model is: Pi Cam IR v2

Network Connection
My motionEyeOS unit is connected to the network via: WiFi

Peripherals
I am using the following peripherals that I consider relevant to this issue:
none

ccrisan, 2018-10-29:
That option only appears in landscape mode because that's where it has effect. On portrait screens (actually *small width screens*), there will always be one single column. That is by design.

closed 2018-10-29 by ccrisan

timotoots, 2020-03-23:
I am sorry to open up this issue, but I think you should reconsider this design choice. Sometimes the MotionEye is embedded in an iframe or a div as in Home Assistant. Then it detects a landscape view and disables columns, but actually it's still a desktop with large screen size where column view would be helpful. I think it's a feature that could be option for users, no need to force design.

starbasessd, 2020-03-23:
Some apps (especially those that read the video streams, and not the web
interface) do that.
Also, try toggling the Fit Frames Vertically option, if available. My
option for Rows goes away if I turn it off.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

simeonovp, 2021-01-01:
I agree with Timtoots. The current functionality also gives me problems. I bought a tablet with FullHD especially for this and run it in portrait mode but only one column is displayed in Motion Eye. Is there a workaround as a solution?

starbasessd, 2021-01-01:
2 things:
1) Try turning on Desktop Version in your browser, if supported. Most
tablet browsers support it, but default to Mobile.
2) A clue here is "HomeAssistant".
https://community.home-assistant.io/t/home-assistant-community-add-on-motioneye/71826
     FRENCK has made changes and won't communicate them.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

simeonovp, 2021-01-01:
@starbasessd : Thanks for your reply!
The solution for me was a small change in the stylesheet file main-css (on my Raspi in /usr/local/lib/python2.7/dist-packages/motioneye/static/css/main.css).
Old:
@media all and (max-width: 1200px)
New:
@media all and (max-width: 800px)
Now I an see the 'Layout Columns' Setting and its works fine with 3 columns.

ljohnso16, 2022-12-05:
Can anyone help me make this change in my HA version of Motioneye? i have 2 ipads i want to use as units but because im inside of HA i cant display 5 cameras on screen at once

starbasessd, 2022-12-05:
It's mobile vs Desktop, and mobile doesn't have the option. If you can force iPad to get the Desktop version you can see the option.

ljohnso16, 2022-12-05:
Not true. I already tried, on multiple browsers, on desktop view i still cannot see the option. this is why i have asked if someone could help me change my CSS file.

xsasx, 2023-01-10:
if you got an solution for this i would like to know it - have the same problem and running motioneye under a docker container