motioneye-project/motioneyeos #1691: Feature request - scheduled uptime (to save battery)

RHTizzy, 2018-10-27:
Requested - An option to shutdown and startup by schedule (similar to motion detection schedule but in this case for the pi itself).

Keep up the good work, this is one of the best setups around!

Chiny91, 2018-10-27:
Not just a software job.   Obviously you could cron job the shutdown but powering up again is not feasible in software.   You could use external circuitry/rtc to power off (say minutes after cron shutdown to be safe) then power back on as you require.   The R Pi doesn't support wake-on-lan either; it is in the chip but not implemented AFAIK.

Maybe possible or maybe chaos would be to cron job a power down whatever uses most power (wifi ?) to save battery power.

I have to ask; what are you not watching during power down ?   It sounds more interesting than the average cam watching for villains.

RHTizzy, 2018-10-29:
Is a cron job something the end user can implement or do you mean you will/can implement shutdown from the frontend? Frontend option would be great but I have no qualms investigating a personal solution if this custom os(?) supports it, all tips welcome! :)  

Startup could indeed be handled by a hardware timer on the power input.

This implementation might be mildly more interesting if you can appreciate that the 'villains' in our use case are cats who only need monitoring in a specific time frame ;p 

Great work btw, can much appreciate all the time and effort that goes into something a slick as this!

Chiny91, 2018-10-29:
I'm just a humble user (so no GUI mods) but [users can implement cron jobs](https://github.com/ccrisan/motioneyeos/wiki/Tweaks#running-commands-at-regular-intervals-cron-support).   I do just that (for remote backup) and it works flawlessly.

Villainous cats... my big problem is spiders webs at night ☹️ although I am cautiously optimistic that #1505  will help one day.   I need to acquire a test R Pi and the Movidius first.

motionEyeOS is indeed slick software.   I'm just back from 2 months away and as usual, 3 cams per Pi on 2 Pi (6 cams total) all ran without a hiccup, allowing me to view remotely, check daily videos... cracking stuff.

RHTizzy, 2018-10-29:
You must have pretty big spiders where you live if you need 6 security cams! ;-p