motioneye-project/motioneyeos #1693: Grey Screen for RTSP Feed (working via other methods)

jstride, 2018-10-27:
I have a hacked dafang camera that works fine if I access it via android directly or via it's own wen interface. It's using RTSP, however it seems to stop working randomly and only a complete docker restart seems to bring it back. I get a grey 'cannot connect image'. It's the second of two cameras I have running in motioneye and it was working fine to start with.

### motionEyeOS Version

I am running motionEyeOS version: (latest docker pull ccrisan/motioneye:master-amd64).

### Board Model

Dell Optiplex

### Camera

DaFang Hacked 

### Network Connection

WiFi

### Log Files

I consider the following log files relevant to this issue:

```
[2:ml2] [ERR] [VID] vid_start: Netcam RTSP failed to open
[2:ml2] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
[4:nc4] [ERR] [NET] netcam_rtsp_handler: Normal resolution: Bad image.  Reconnecting with camera....
[2:ml2] [ERR] [NET] netcam_rtsp_open_context: Normal resolution: Unable to open camera((null)): Immediate exit requested
```

jstride, 2018-10-27:
Further investigation seems to be that this is based on which ever camera was last added. i.e. if I delete and re-add the camera the second one works and the first times out. Load doesn't seem bad?

`top - 07:13:33 up 8 min,  0 users,  load average: 0.30, 0.33, 0.20
Tasks:   6 total,   1 running,   5 sleeping,   0 stopped,   0 zombie
%Cpu(s):  5.5 us,  0.4 sy,  0.0 ni, 93.0 id,  0.2 wa,  0.0 hi,  0.9 si,  0.0 st
KiB Mem :  8044372 total,  5663360 free,  1478680 used,   902332 buff/cache
KiB Swap:  4194300 total,  4194300 free,        0 used.  6300988 avail Mem

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND
  189 root      20   0  847440 123968  30896 S  21.3  1.5   0:50.32 motion
    8 root      20   0  393424  35652  11308 S   0.7  0.4   0:06.32 meyectl
  258 root      20   0   43296   3332   2832 R   0.3  0.0   0:00.01 top
    1 root      20   0    4592    812    748 S   0.0  0.0   0:00.02 sh
   22 root      20   0  165420  21988   2640 S   0.0  0.3   0:00.00 meyectl
  247 root      20   0   18492   3496   3052 S   0.0  0.0   0:00.01 bash
`

ccrisan, 2018-10-29:
@jstride you're probably running motionEye, not motionEyeOS. If you're not getting streaming image from Motion, please [report a Motion issue](https://github.com/ccrisan/motioneye/wiki/Report-Motion-Issue).

closed 2018-10-29 by ccrisan