motioneye-project/motioneyeos #1695: Broken streaming URL for 3rd part apps

RHTizzy, 2018-10-29:
After updating to the latest motioneyeos firmware, the video streams are no longer accessible via the streaming URL to 3rd party apps.  Is it possible to identify what was changed in the firmware so 3rd party vendors can make changes to their software?

Preliminary Docs
I confirm that I have read the CONTRIBUTING guide before opening this issue.

I confirm that I have read the FAQ before opening this issue.

motionEyeOS Version
I am running motionEyeOS version: 0.39.3

Board Model
Pi Zero W

Camera
I am using the following type of camera: MMAL

My camera model is: Pi Cam IR v2

Network Connection
My motionEyeOS unit is connected to the network via: WiFi

Peripherals
I am using the following peripherals that I consider relevant to this issue:
none

Log Files

ccrisan, 2018-10-31:
The streaming URL hasn't changed at all in the past one or two years, so I don't think that's the issue. Nor has the authentication mechanism. What are these 3rd party apps that don't work and what is the URL you've been using?

RHTizzy, 2018-11-01:
I have 6 cams which until recently were all working fine using the IP cam viewer app (by Robert Chou). 3 of the pi-cams were updated to the latest motionEyeOs firmware and 3 are still using the previous firmware. The IP cam viewer app was configured for all 6 before updating the firmware on the cams. After updating the motionEyeOs firmware, the 3 cams are no longer reachable using the app. Configuration is identical for all 6 using the webGUI. The webGUI works fine for all 6 cams. The URL's used by the app is the one provided by the webGUI, ie 'http://nn.nn.nn.nnn:pppp"

ccrisan, 2018-11-05:
If the broken apps are represented by the "IP cam viewer", please report this issue to them.

closed 2018-11-05 by ccrisan

ccrisan, 2018-11-05:
@RHTizzy you have your answer in https://github.com/ccrisan/motioneyeos/issues/1707.