motioneye-project/motioneyeos #1698: Pre Release not working

mehtaman, 2018-10-30:
### Preliminary Docs
I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.
**YES**

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.
**YES**

### motionEyeOS Version
I am running motionEyeOS version: **20180627**.

### Board Model
I am using the following board/model: **Raspberry PI 3B+**

### Camera
I am using the following type of camera: **V4L2**
My camera model is: **Pi Camera**

### Network Connection
My motionEyeOS unit is connected to the network via: **WiFi**

### Peripherals
I am using the following peripherals that I consider relevant to this issue:
**None**

### Log Files
I consider the following log files relevant to this issue:
 **N/A**
 
### Issue
Since the builds have changed to nightly-dev my MotionEyeOS has not prompted for an update since 20180627 but under Expert Settings > Enable Prereleases is set to ON.

### Request
Can this be fixed or can an option for Nightly builds be added as an option?

Thank you

ccrisan, 2018-11-05:
We don't want to make it *that* easy for users to update to nightly builds; it's too risky. If you want to update your system to the latest nightly release, just do a `fwupdate upgrade nightly-dev` via SSH and... fingers crossed :)

closed 2018-11-05 by ccrisan

mehtaman, 2018-11-06:
perfect, thank you for the command 👍