motioneye-project/motioneyeos #1705: movie_codec parameter not accepted by motion

afeno, 2018-11-03:
### More details
I added the following parameters in the "Extra Motion Options" fields:
movie_codec mp4
timelapse_fps 30
timelapse_mode daily
timelapse_interval 1

The last three parameters are accepted and working fine. However, the movide_codec is not recognized. See the logs below.

**Why the movie_codec mp4 is not accepted/unknown?**

### motionEyeOS Version

I am running motionEyeOS version: motionEyeOS 20180627

### Board Model

Raspberry PI 3B+

### Camera

I am using the following type of camera: Network Camera

My camera model is: VstarCam C7186WIP

### Network Connection

My motionEyeOS unit is connected to the network via:  Ethernet

### Peripherals

I am using the following peripherals that I consider relevant to this issue: Nothing

### Log Files

I consider the following log files relevant to this issue: motioneye.log
```
[1994240896:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
[1994240896:motion] [ALR] [ALL] conf_cmdparse: Deprecated config option "thread" since after version 3.4.1:
[1994240896:motion] [ALR] [ALL] conf_cmdparse: The "thread" option has been replaced by the "camera" option.
[1994240896:motion] [NTC] [ALL] config_camera: Processing camera config file thread-1.conf
[1994240896:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "movie_codec"
[1994240896:motion] [NTC] [ALL] motion_startup: Motion 4.1.1 Started
[1994240896:motion] [NTC] [ALL] motion_startup: Logging to syslog
[1994240896:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[1994240896:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
```

jasaw, 2018-11-03:
`ffmpeg_video_codec` was renamed to `movie_codec` in motion release 4.2, so if you're running older motion version, you may have to use `ffmpeg_video_codec`.

afeno, 2018-11-03:
Thanks,
Yes, I'm running 4.1.1 but `ffmpeg_video_codec` is rejected as an "Extra motion Option" parameter. When I add the line and then I Apply the changes in motioneye, it the line is just deleted.

jasaw, 2018-11-03:
I think that's because MotionEye recognizes `ffmpeg_video_codec` option and removes it from the "Extra motion option" list. You may have to SSH into your RPi and add `ffmpeg_video_codec` option by hand. Give that a try.

afeno, 2018-11-03:
I found the answer
https://motion-project.github.io/motion_config.html#timelapse_codec
:(
the timelapse will never be in mp4 format.

closed 2018-11-03 by afeno