motioneye-project/motioneyeos #1712: Why is FTP so bad?

talormanda, 2018-11-12:
Every time I connect to it, it seems to time out, or take forever to load. All I am doing is loading a directory or moving a few files and it will stop connecting. SMB seems to be perfect though.

ccrisan, 2018-11-13:
That's not normal. It's hard to say what's going on there with your FTP server but it should definitely not say "connection refused". Please attach all the log files and maybe we'll be able to identify the problem.

talormanda, 2018-11-13:
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/2577368/dmesg.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/2577369/boot.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/2577370/messages.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/2577371/motioneye.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/2577372/motion.log)

@ccrisan I transferred 193 files (pictures) and it randomly gave the same ERROR, then kept going a little while later. Sometimes it will do this, other times it will ERROR and refuse to connect again. This has been happening always, even between reloads of the OS and switch from a 3b to a 3b+.

ccrisan, 2018-11-16:
@talormanda you're having some network connectivity issues:

    user.notice netwatch: cannot connect to www.google.com:80, rebooting

and

    user.notice ethernet: disconnected

You have enabled the network connectivity watching option, which reboots your system when it detects that connection with the given host cannot be established. This is why you're getting the FTP errors.

closed 2018-11-16 by ccrisan

talormanda, 2018-11-16:
@ccrisan I will test this and report back.

talormanda, 2018-11-16:
@ccrisan how would this be a network connectivity issue? My pi is hardwired directly to the router?