motioneye-project/motioneyeos #1713: Motion Detection Recording lots of small videos of various lengths and end early

xomp, 2018-11-13:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motionEyeOS dev20181008.

### Board Model

I am using the following board/model: Raspberry PI 3B.

### Camera

I am using the following type of camera: MMAL

My camera model is: https://www.amazon.com/gp/product/B07DNSKMZ1/ref=oh_aui_detailpage_o04_s00?ie=UTF8&psc=1

### Network Connection

My motionEyeOS unit is connected to the network via: Wifi

### Peripherals

### Log Files

### Movies Settings
Using Matroska Video, Motion Triggered, 0 seconds Maximum Movie Length, Preserve Movies: Forever.
 
### Motion Detection Settings
1.5% Frame Change Threshold
Auto Noise Detection: On
Light Switch Detection: 0%
Despeckle Filter: Off
Motion Gap: 5 secs
Capture Before: 5 secs
Captured After: 5 secs
Minimum Motion Frames: 5 secs
Mask: On
Mask Type: Smart
Smart Mask Sluggishness: 5
Show Frame Changes: Off
Create Debug Media Files: Off

### The Problem I'm having
Recordings begin to work once motion is detected, though for some reason, they either end shorter than supposed to or I end up with 50+ small 160kb Movies on the SDCard recording portions of the same event. Then sometimes things work fine and the even is recorded in its entirety.

I have tried so many times to get this to play nice but it's simply refusing to comply. Anyone else having this issue or am I #blessed to be the first?

Thank you for your time.

jasaw, 2018-11-13:
Your `motion gap` is too small, which means a new video will be created after 5 seconds of no motion.
Your `captured after` is also too small. Do you actually mean 5 seconds or 5 frames? On MotionEye web interface, this field has a unit of "frames" rather than "seconds". 5 frames captured after means you'll end up with choppy video. Try increasing it to something more sensible, like 10 seconds (10 x frame rate).

I suggest you turn on `show frame changes` while you tweak your motion detection settings. It shows you where motion is detected and the number of pixels that have detectable change (will be compared against `frame change threshold`).

xomp, 2018-11-13:
> Your `motion gap` is too small, which means a new video will be created after 5 seconds of no motion.
> Your `captured after` is also too small. Do you actually mean 5 seconds or 5 frames? On MotionEye web interface, this field has a unit of "frames" rather than "seconds". 5 frames captured after means you'll end up with choppy video. Try increasing it to something more sensible, like 10 seconds (10 x frame rate).
> 
> I suggest you turn on `show frame changes` while you tweak your motion detection settings. It shows you where motion is detected and the number of pixels that have detectable change (will be compared against `frame change threshold`).
> 
> If you get motion detection from wind blowing on trees, try turning on `despeckle filter` to reduce the false positives.

Thank you for your quick response!

I've update the figures on the settings you pointed out and will give it a try tomorrow to see if things improve. I've upped my motion gap to 15sec, Capture Before to 15 frames, Capture After to 15 frames, & Show Frame Changes is now on. I'm guessing I'm having trouble understanding how these all correlate with the performance of all things involved based off the tiny amount of info available in the tooltip blurb. I plan on reading over some of the motion documentation later tonight to see if I can gather a better understanding.

Again, thanks for your help!

xomp, 2018-11-15:
Okay, after much testing. I've managed to get better results. Still have an issue where the recordings end early while motion is still going on. Not sure what is causing that specifically, but it's bloody well annoying. I'll peruse the motion FAQ to see if I can get a better grasp of all these settings and maybe have a try at making more adjustments. 

So far, I'm very pleased with this little setup. Motioneyeos is so good. Thanks for all the hard work you pour into it. It shows!

xomp, 2018-11-16:
Okay, checking in. After hours and hours of testing to no avail and using the default settings. I have determined that the motion portion of this software either isn't working to spec, or the configuration needed to make it functional is too advanced. I still can't seem to get past the issue of the recordings getting cut off too early no matter what I try. I was hoping to use my pi for something more than a Retropie arcade, but I think I'm gonna have to go back to that instead of using it for camera purposes. At any rate, thanks all for your help and good luck on your project!

jasaw, 2018-11-21:
If you want further help, you could attach a short video file that shows the cutting short problem. Make sure the "show debug frames" is turned on so we can see the red box on detected motion area.

closed 2018-12-05 by ccrisan

mhaddy, 2018-12-20:
@xomp Can you post your current settings? I'm having a staggering amount of false positives and trying to tune.