motioneye-project/motioneyeos #1714: Motioneye unable to connect to network camera

TFCDRDEATH, 2018-11-14:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
 

http://192.168.1.192/web/index.html

TFCDRDEATH, 2018-11-14:
Unable to connect network camera to motion pi

http://192.168.1.192/web/index.html

Say not a compatible network camera but doesn't say why?

ccrisan, 2018-11-16:
I believe that by "motion pi" you mean motionEyeOS running on a Raspberry PI. In that case, you'll need to find the correct streaming URL (MJPEG or RTSP) for your camera. I'm 99% positive that something ending with an `index.html` is not a valid streaming URL, but rather a webpage.

closed 2018-11-16 by ccrisan