motioneye-project/motioneyeos #1715: No network created

AlexTollola, 2018-11-15:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

AlexTollola, 2018-11-15:
Hi, I´ve already included the dnsmasq.conf and the hostapd.conf files in the boot partition. I also modified the os_networkless configuration and set it to true. But after initializing my raspberry it won´t create the network.

ccrisan, 2018-11-16:
@AlexTollola please attach your log files (boot.log, dmesg.log and messages.log). If you can't get to them, attaching a monitor could help you see what's going on there.

AlexTollola, 2018-11-16:
Here are the files.
I was able to create the network, but shortly after it was created it
disapperared.

ccrisan, 2018-11-16:
@AlexTollola you did not attach any files. Please **do not reply** to the email directly; it won't work that way.

AlexTollola, 2018-11-16:
[boot.txt](https://github.com/ccrisan/motioneyeos/files/2591028/boot.txt)
[dmesg.txt](https://github.com/ccrisan/motioneyeos/files/2591029/dmesg.txt)
[dnsmasq.txt](https://github.com/ccrisan/motioneyeos/files/2591030/dnsmasq.txt)
[hostapd.txt](https://github.com/ccrisan/motioneyeos/files/2591031/hostapd.txt)
[messages.txt](https://github.com/ccrisan/motioneyeos/files/2591032/messages.txt)