motioneye-project/motioneyeos #1717: ### Board Model  I am using the following board/model: Raspberry PI 3 A+

freeflocw, 2018-11-17:
### Board Model  I am using the following board/model: Raspberry PI 3 A+

I tried for the new Raspi 3A+ both version of Motioneyeos for PiA  and Pi3+.
Non worked!
Any idea?

RPiSpy, 2018-11-17:
I've just got a Pi 3 Model A+. Keen to get MotionEyeOS working but I suspect it will take the developers some time to get their hands on a new Pi and make the required updates. There is no reason it shouldn't work as the 3A+ is almost identical to the 3B+.

freeflocw, 2018-11-18:
I can send a A+

ccrisan, 2018-11-18:
I am working on updating the OS. The kernel and firmware for the RPi boards will also get updated. You'll be able to try the new OS image as soon as I release it. I personally don't own an 3A+ but I don't think that's going to be necessary, for now.

KeithMacDonald, 2018-11-21:
Thanks, that's good news. :-)
I too have just got a Pi 3 Model A+, and am keen to get MotionEyeOS running on it.
I look forward to seeing the new release.

iwbnwif, 2018-11-21:
Great news, I'm also keen to try this on a Pi 3 A+.

freeflocw, 2018-11-21:
Optimal for ipcam, performance and size. What would be a good case for the 3A+ with camera housing like the original zero case?

KeithMacDonald, 2018-11-22:
Pi 3a+ case with space for camera?
Try this: https://thepihut.com/products/adafruit-raspberry-pi-a-case

freeflocw, 2018-11-22:
yes, thanks, i have seen it. but simple white or black.
there will be also a Org.Pi case coming december, but not for camera like 3b+
https://www.welectron.com/Offizielles-Raspberry-Pi-3-A-Gehaeuse-rot-weiss

freeflocw, 2018-11-26:
Great, thank you very much. crisan.  On my 3A+ it's working.,great job in total.

ccrisan, 2018-11-26:
Good to know. Thank you @freeflocw for reporting back!

closed 2018-11-26 by ccrisan

joland, 2019-01-18:
@ccrisan So which version is need? I just download motioneyeos-raspberrypi3-20181209.img.xz 
and installed it for the Raspberry pi model 3 A+. But that one did not boot.

freeflocw, 2019-01-18:
I got on several 3A+ with this version 3 running

joland, 2019-01-18:
@freeflocw i swapped SD with a Stretch install for the 3B+ and it loaded without a hitch. Booting with motioneyeos-raspberrypi3-20181209.img it get stuck right from the start. It only shows the color array screen. Are there any special installation settings i should be aware of?

joland, 2019-01-18:
I mixed up version pi3-20181209 with pi3-dev20190118. The latter doesn't work with a pi 3 A+ but first does.

ccrisan, 2019-01-18:
@joland I'll try to reproduce the problem myself and get back to you.

freeflocw, 2019-01-30:
dev20190128 works on my 3A+ too.