motioneye-project/motioneyeos #1719: Network load causes kernel stack dump then reboot

m00k12, 2018-11-18:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20180627

### Board Model

I am using the following board/model: RPi Zero W

### Camera

I am using the following type of camera: MMAL Camera

My camera model is: 5MP camera fromthe OctoCam kit

https://shop.pimoroni.com/products/octocam-pi-zero-w-project-kit

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

* none

### Log Files

I consider the following log relevant to this issue:
```

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.30 (root@281332b64809) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1 Thu Jun 28 20:05:44 UTC 2018
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Zero W Rev 1.1
[    0.000000] Memory policy: Data cache writeback
[    0.000000] cma: Reserved 8 MiB at 0x17400000
[    0.000000] On node 0 totalpages: 98304
[    0.000000] free_area_init_node: node 0, pgdat c09a263c, node_mem_map d7c8a500
[    0.000000]   Normal zone: 864 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 98304 pages, LIFO batch:31
[    0.000000] random: fast init done
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 97440
[    0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:48:E7:67 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.fiq_fix_enable=1 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait ro rootflags=noload panic=10 quiet loglevel=1 ipv6.disable=1
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 370420K/393216K available (6349K kernel code, 562K rwdata, 1968K rodata, 440K init, 677K bss, 14604K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] ftrace: allocating 23857 entries in 70 pages
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000029] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000059] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000145] bcm2835: system timer (irq = 27)
[    0.000690] Console: colour dummy device 80x30
[    0.000715] console [tty1] enabled
[    0.000750] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.060296] pid_max: default: 32768 minimum: 301
[    0.060800] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[... 198 lines cut ...]
[  157.362725] 7ee0: 00000000 00000000 00000040 b5d591a0 d5287fac d5287f00 c0009200 c0639bf0
[  157.368223] 7f00: 00000001 00000003 d52bf460 00000000 00000000 00000000 00000000 00000000
[  157.373725] 7f20: 00000000 00010000 d52bf468 00000001 00000000 d4dbd638 d5286000 00000000
[  157.379178] 7f40: d5287f7c d5287f50 c015a42c c019d324 00000000 c0177dec 00000000 d52bf460
[  157.384648] 7f60: d5287f7c c015a970 00000000 d52bf460 d5287fa4 b4ba5410 01877050 b6f0e548
[  157.390141] 7f80: d5287fa4 d5287f90 c050db88 c050da7c 00000000 00000000 00000000 d5287fa8
[  157.395589] 7fa0: c000fe00 c050db6c b4ba5410 01877050 00000010 01b384d8 0001012a 00000000
[  157.401067] 7fc0: b4ba5410 01877050 b6f0e548 00000121 00000001 01b384d8 0001012a beca8b3c
[  157.406556] 7fe0: 00000000 beca8b20 beca8afc b6d70154 80000010 00000010 17ffa861 17ffac61
[  157.412031] [<c0580d80>] (tcp_push) from [<c058422c>] (tcp_sendmsg_locked+0x2dc/0xe4c)
[  157.417526] [<c058422c>] (tcp_sendmsg_locked) from [<c0584dd4>] (tcp_sendmsg+0x38/0x4c)
[  157.423050] [<c0584dd4>] (tcp_sendmsg) from [<c05afd00>] (inet_sendmsg+0x44/0x78)
[  157.428534] [<c05afd00>] (inet_sendmsg) from [<c050ca18>] (sock_sendmsg+0x24/0x34)
[  157.434041] [<c050ca18>] (sock_sendmsg) from [<c050db38>] (SyS_sendto+0xc8/0xf0)
[  157.439790] [<c050db38>] (SyS_sendto) from [<c050db88>] (SyS_send+0x28/0x30)
[  157.442675] [<c050db88>] (SyS_send) from [<c000fe00>] (ret_fast_syscall+0x0/0x28)
[  157.448611] Code: e15c0004 03a0c000 e2115902 1a000024 (e5dce028) 
[  157.451565] ---[ end trace b2190856e28f1cba ]---
[root@meye-7348e767 ~]# Connection to 192.168.1.95 closed by remote host.

```
 
Shortly after this, the RPi will reboot.

I'm currently using a branded 1.2A power supply which should be enough for a Pi Zero W, but will try a 2.5A official PSU next.

m00k12, 2018-11-18:
Same stack trace on a different Pi when streaming media with Plex:

https://www.reddit.com/r/raspberry_pi/comments/8bvvkx/plex_on_pi_3b_has_been_working_for_monthsyears/

m00k12, 2018-11-18:
I found this: https://patchwork.ozlabs.org/patch/886324/

Apparently the patch is/was queued for 4.14.32.  This version of motioneyeOS is running 4.14.30.

Time for a new kernel (4.14.81 exists now, for example)?

ccrisan, 2018-11-18:
The release that's cooking as we speak will run 4.14.81.

m00k12, 2018-11-19:
> The release that's cooking as we speak will run 4.14.81.

Great!  Is that in the nightlies yet?  If so, would you recommend I try that (or might I find other issues)?

zapletaldavid, 2018-11-21:
I've got the same issue (Unable to handle kernel NULL pointer dereference at virtual address 00000028) on RPi3, rebooting over and over. Same on latest nightly version. Finally installed Raspbian (kernel 4.14.79) yesterday, MotionEye running w/o problems since then. Looking forward to new version, I need just the camera features w/o any garbage around and this is just excellent load&go package for amateurs like I am :-)

ccrisan, 2018-11-26:
The latest nightly (dev20181126) is available. It runs the kernel & firmware currently present in Raspbian.

You'll need to manually write the OS image from scratch, as the partition sizes have increased and the firmware update process will fail.

bsimmo, 2018-12-16:
Testing the 5G on the Pi3B+ I get this trace, so it has to be the unstable/testing versions to stop this ?

When is testing possibly going stable..
any chance of a firmware bump (and 3A+ inclusion) for the old current setup ?
Save me some faff ;-)

ccrisan, 2018-12-24:
@bsimmo what version are you referring to? The current prerelease (20181209) has the latest firmware available at that time, which includes 3A+.

bsimmo, 2018-12-24:
current 'release' version, hence the question to see if the current version can have a bump to save an SD rewrite.
I'm not going to faff with the pre-release as it's needs a rewrite of the SD :-1:

m00k12, 2018-12-28:
I've been running 20181209 for a while now and not seen any of the Pi Zero W's network lock up, so closing this issue.

closed 2018-12-28 by m00k12

tnmdo, 2019-06-28:
I'm running the latest 20190427 on Raspi Zero W and I am encoutering this issue permanently :(

jasaw, 2019-07-01:
@tnmdo Maybe it's related to this issue? https://github.com/ccrisan/motioneyeos/issues/1945

tnmdo, 2019-07-01:
@jasaw Yes, maybe. Thank you very much for pointing out. I only thought about the TCP/IP stack, but didn't knew that ticket.

I've running some pieces of newer revisions of the Pi Zero W - on the first one I got it was running for days without an outtake, but in the last days I could not get any device up and running for > 1 min.

Perhaps it's the hardware capability - and got worse over here because of warm weather. Doesn't matter - it's a pitty. I have about 8 Zero's now that all have to be replaced with Raspi 3 or newer, just in case.

But hey, that's IT. Thanks again.

tnmdo, 2019-07-01:
@jasaw I think it's solved via overclocking and forcing turbo mode. Seems to have excellent performance now with low latency.

@ccrisan Thank you very much for this wonderful piece of software.