motioneye-project/motioneyeos #1722: 3B+ 5GHz will not join

raymate, 2018-11-25:
Setup fresh build of motioneye, used ethernet of course on my new 3B+

All set to go with latest stable build, setup wifi to my 5GHz network and it will not join it, 

during boot get wpa_supplicant failed (brcmfmac) error (if that helps)

Tried it with my 2.4GHz and this work. (I have different SSID setup for 2.4 and 5)
But I really want to get the camera onto my 5GHz network. Is this a known issue?

Anything I can try?

Thanks

ccrisan, 2018-11-26:
Does it work with Raspbian and your WiFi setup?

raymate, 2018-11-26:
Yes works fine under Raspbian on the 5GHz side

ccrisan, 2018-11-26:
@raymate can you please try the latest nightly build?

raymate, 2018-11-26:
Just tried the dev20181126 build, no luck

Still gives same wpa error when starting WPA

ccrisan, 2018-11-26:
Are we talking about the exact same `wpa_supplicant.conf`?

raymate, 2018-11-26:
I’m referring to when you boot the Pi I have a monitor plugged in to easily see the IP the Pi gets. It’s on the boot screen as it starts up

You get the list of the Pi doing stuff it’s during this point I see the error

Ray

ccrisan, 2018-11-26:
@raymate I get that and that's fine. What I am asking if you're using the same `wpa_supplicant.conf` file for both motionEyeOS and Raspbian tests.

raymate, 2018-11-26:
Ok,  after doing some google I see what you mean. I can use the same file and copy it. Didn’t even know that was possible :)

To answer question no I have not used the same file 

I’m at work. Is so when I get home I will try and copy the file from the Pi that is connected to the 5ghz and move that to the motioneye Pi 

I shall report back

raymate, 2018-11-26:
Ok so I found wpa file on the motioneye under the Etc folder is that correct location

It contains no ssid or password info just a line saying ssid scan_ssid=1

On my Raspbian Pi I found the same file with the ssid and password for my 5ghz network

I can’t work out how to copy the file over 

I even typed the ssid and password lines into the file on motioneye but it would not save said read only

Ray

raymate, 2018-11-27:
Its looking like that is not the location of the wpa_supplicant.conf file, I put the wifi setting in motion eye for my 2.4Ghz and that joins straight away and that file I found contains no info just he scan_ssid command

Where do I find that file?

Jaiksmith, 2018-11-27:
I just did a build, and I've got the same issue, exactly the same issue. We're both running 2.4ghz & 5ghz wifi, probably both on comcast with the same router. 
I tested the motioneye image on my iPhone hotspot and it connected on the first try, it also connects to 2.4ghz without a problem. For simplicity, I have set the password the same, so only the SSID changes. My two SSIDs are:
XF-2.4ghz
XF-5ghz

It's extremely confusing that it won't connect to one, but it will the other. I'm in the process of setting up a Raspbian Stretch image to see if I can pull the wpa_supplicant file as suggested.

Jaiksmith, 2018-11-27:
On the initial setup of Raspbian Stretch after passing the locale stage and searching for wifi Raspbian finds the 2.4ghz network, but not the 5ghz network. I'll see if I can add it in manually. Both are set for discovery within the router.

raymate, 2018-11-27:
Pleased I’m not alone. Most people don’t split the 2.4 and 5 ssid so I guess most people have not noticed it’s going to the 2.4

I specifically want to get my pis of the 2.4 and onto the 5.  I do have some wired bit not all 

Got the file from stretch but can’t find it’s location on the motioneye SD card

Jaiksmith, 2018-11-27:
So, I cannot get raspbian to connect to the 5ghz network at all. I've even attempted to change the name of the network, and it's security.  I've also forced the wpa_supplicant.conf to have the correct ssid & psk. This still would not allow it to connect. I have added the wpa_supplicant.conf in two ways, one was copying it over, the second was using the [WIFI preconfiguration instructions.](https://github.com/ccrisan/motioneyeos/wiki/Wifi-Preconfiguration) Neither of those worked. I'll use the 2.4ghz network for now...Any help would be appreciated though.

Jaiksmith, 2018-11-28:
raymate,
 I've discovered my issue...I've got a RPi 3B....not the 3B+. 3B does not support 5ghz unless there's a USB dongle/antenna that supports it. 3B+ should support 5ghz, but I'm not sure if the standalone builds will. If possible, I would suggest building up the latest stable Stretch then manually installing MotionEye. This may resolve your issue.

raymate, 2018-11-28:
Thanks Jaik for the update. 

Never done an install from stretch before. Just used the motioneye pre made image 

I shall try it

I also have a USB dongle that is 5GHz perhaps I should try that also. 

Cheers

bsimmo, 2018-12-15:
3B+ and 3A+ on Raspbian need the country code in the OS set first, I don't know if this is a software or a firmware/hardware setup.
This is requirement of of the 5GHz regulations.

I don't know how motioneyeOS solves this.

bsimmo, 2018-12-16:
As a check, edit the file to use this format (note the order of the top three lines)
possibly try GB first, then alter as need for your country. 
```
ctrl_interface=/var/run/wpa_supplicant
update_config=1
country=GB

network={
	ssid="myssid"
	psk="mypasscode"
	key_mgmt=WPA-PSK
}

```

JubilantReaper, 2019-01-20:
bismmo - Great info, the challenge was the path to locate it, I updated /data/etc/wpa_supplicant.conf to include the country and updated the SSID, rebooted and BOOM, on 5GHz... Maybe ccrisan can add CountyCode in setup somewhere...

raymate, 2019-01-20:
> bismmo - Great info, the challenge was the path to locate it, I updated /data/etc/wpa_supplicant.conf to include the country and updated the SSID, rebooted and BOOM, on 5GHz... Maybe ccrisan can add CountyCode in setup somewhere...

This is good news, I sort of give up as I couldn't find the file to edit. So lets see if I can figure this out

raymate, 2019-01-20:
> > bismmo - Great info, the challenge was the path to locate it, I updated /data/etc/wpa_supplicant.conf to include the country and updated the SSID, rebooted and BOOM, on 5GHz... Maybe ccrisan can add CountyCode in setup somewhere...
> 
> Found it, added country code but no luck

Now works, I created a fresh install from a new image and setup fresh. Now running on 5GHz.

@ccrisan when get time please ass the country code, we can then enjoy full 5GHz with the newer Pi's thanks

Thanks @bsimmo for showing the location of the file. Works like a charm

jasaw, 2019-01-20:
@raymate Country code has always been in the [wifi preconfiguration wiki](https://github.com/ccrisan/motioneyeos/wiki/Wifi-Preconfiguration), step 11.

AIexBV, 2019-03-10:
I had this issue as well. The solution for me was to change the WiFi channel from above 100 to below 50.