motioneye-project/motioneyeos #1725: TL-WN721N fails to connect wireless

Vants, 2018-12-01:
Hello!

I installed motioneyeos-raspberrypi-20180627 to my Raspberry Pi Zero 1.3. I use WiFi dongle TL-WN721N to get Pi to network. 

When I boot to MotionEye I get "**wpa_supplicant: failed (ath9k_htc)**". I also tried other WiFI dongle and still same problem (id was different).

Both dongles worked in Raspian.

My wpa_supplicant.conf (UNIX endings and UTF-8)
```
ctrl_interface=/var/run/wpa_supplicant
update_config=1
country=ee

network={
scan_ssid=1
ssid="Test"
psk="12345678"
}
```

ccrisan, 2018-12-03:
@Vants can you please try the latest 20181129 version?

Vants, 2018-12-16:
Tried version 20181129. No wpa_supplicant error anymore.

Thanks!

closed 2018-12-16 by Vants