motioneye-project/motioneyeos #1726: System of frozen or unreachable

Klausoleum, 2018-12-01:
My MotionEyeOs system (RPi 3B +) is more often unreachable. Last time every other day.

System design:
- MotionEyeOs as motion detection server, Odroid HC1, works great
- MotcionEyeOS as a webcam on RasperryPi zero W, 800x600px resolution, works great
**- MotionEyeOS as webcam on RaspberryPi 3B+**, 1920x1080px resolution, connected via Wi-Fi with good signal strength, 2.4Ghz 80Mbps, **often freezes or is not available**

last time I restarted the system at night (cron), no improvement

I have activated the debugging mode for detailed log files

Attached the log files. I could not find any reason why the system is sometimes unreachable.
Last time the system was no longer available on 1.12.2018 13:41
Restarted on 1.12.2018 at 22 o'clock

Can someone on the basis of my log files recognize why the system is no longer available?
[boot.log](https://github.com/ccrisan/motioneyeos/files/2636347/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/2636348/dmesg.log)
[Uploading messages.log…]()
[Uploading motion.log…]()
[Uploading motioneye.log…]()

m00k12, 2018-12-08:
Looks like you are running kennel 4.14.30 on your Pi.

I would guess it is the same issue as in #1719.  

You could try the nightly build with the newer kernel instead.

m00k12, 2019-01-26:
To follow up, 20190119 is running perfectly with motion detection on my two pi zero w's.

Prior releases either locked up all networking (except ping), it the camera feed stopped.

20190119 has been all good running non-stop so far.