motioneye-project/motioneyeos #1727: Unable to launch Fast Network Camera on latest dev build.

smithatlanta, 2018-12-02:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

dev20181130

### Board Model

Raspberry Pi Zero W

### Camera

Fast Network Camera

Raspberry Pi Camera Module V2-8 Megapixel,1080p

### Network Connection

Wifi

### Peripherals

No peripherals.

### Log Files

This happens during the reboot after I have turned on Advanced Settings and then turn on Fast Network Camera.  It just loops.  I'm going to see if I can find a way to provide more information.

I get all the way to..

...
* starting streameye: done
* starting motioneye: failed

ccrisan, 2018-12-03:
Interesting. I'll need to check this and see what's going on. Thanks for reporting.

jasaw, 2018-12-05:
I can reproduce this bug too. MotionEye hangs when started from `/etc/init.d/S85motioneye` script, but works fine when I run it manually like so `meyectl startserver -c /data/etc/motioneye.conf`

ccrisan, 2018-12-09:
The problem arises from the fact that Tornado recently started to check the IOLoop's initial pid against the current process' pid and this may in some cases break the daemonizing/forking mechanism implemented in motionEye.

This happens only when using FNC because only then the `streameyectl.py` module is imported and the IOLoop is then initialized early.

The solution is to reset the IOLoop's pid before starting the main loop.

Fixed in https://github.com/ccrisan/motioneye/commit/d77d3d81286e9fe4d25800d19df730f59c52e6e1.

closed 2018-12-09 by ccrisan