motioneye-project/motioneyeos #1728: No video on attached Raspberry Pi 3 A+

rockman55, 2018-12-02:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version 

I am running motionEyeOS version: (motionEyeOS 20180627, ver 4.1.1motionEye Version | 0.39.2).

### Board Model

I am using the following board/model: (Raspberry PI 3 Model A+).

### Camera

I am using the following type of camera: (Simple MJPEG Camera, ).

My camera model is: (Raspberry Pi Official v1).

### Network Connection

My motionEyeOS unit is connected to the network via: ( Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 
 * (e.g. USB External Hard Disk, for file storage)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
2018-12-02 03:17:21: [motioneye]  WARNING: mjpg client connection for camera 7 on port 8087 is closed
2018-12-02 03:27:22: [motioneye]  WARNING: mjpg client connection for camera 7 on port 8087 is closed
2018-12-02 03:29:12: [motioneye]  WARNING: mjpg client connection for camera 1 on port 8081 is closed
2018-12-02 03:29:42: [motioneye]  WARNING: mjpg client connection for camera 6 on port 8086 is closed
2018-12-02 03:38:23: [motioneye]  WARNING: mjpg client connection for camera 7 on port 8087 is closed
2018-12-02 03:39:33: [motioneye]  WARNING: mjpg client connection for camera 7 on port 8087 is closed
2018-12-02 03:40:43: [motioneye]  WARNING: mjpg client connection for camera 7 on port 8087 is closed
2018-12-02 03:40:43: [motioneye]  WARNING: mjpg client connection for camera 9 on port 8089 is closed
2018-12-02 03:42:33: [motioneye]    ERROR: failed to cleanup media files: [Errno 30] Read-only file system: '/data/media/sda1/office/.keep'
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/cleanup.py", line 86, in _do_cleanup
  File "/usr/lib/python2.7/site-packages/motioneye/mediafiles.py", line 293, in cleanup_media
IOError: [Errno 30] Read-only file system: '/data/media/sda1/office/.keep'
2018-12-02 03:48:54: [motioneye]  WARNING: mjpg client connection for camera 9 on port 8089 is closed
2018-12-02 03:52:34: [motioneye]  WARNING: mjpg client connection for camera 3 on port 8083 is closed
2018-12-02 03:58:14: [motioneye]  WARNING: mjpg client connection for camera 9 on port 8089 is closed
2018-12-02 04:03:35: [motioneye]  WARNING: mjpg client connection for camera 2 on port 8082 is closed
2018-12-02 04:04:05: [motioneye]  WARNING: mjpg client connection for camera 9 on port 8089 is closed
2018-12-02 04:07:35: [motioneye]  WARNING: mjpg client connection for camera 9 on port 8089 is closed
2018-12-02 04:12:15: [motioneye]  WARNING: mjpg client connection for camera 9 on port 8089 is closed
2018-12-02 04:14:35: [motioneye]  WARNING: mjpg client connection for camera 7 on port 8087 is closed
2018-12-02 04:15:45: [motioneye]  WARNING: mjpg client connection for camera 7 on port 8087 is closed
2018-12-02 04:16:56: [motioneye]  WARNING: mjpg client connection for camera 7 on port 8087 is closed
2018-12-02 04:18:06: [motioneye]  WARNING: mjpg client connection for camera 9 on port 8089 is closed
2018-12-02 04:19:16: [motioneye]  WARNING: mjpg client connection for camera 7 on port 8087 is closed
2018-12-02 04:21:36: [motioneye]  WARNING: mjpg client connection for camera 7 on port 8087 is closed
2018-12-02 04:21:36: [motioneye]  WARNING: mjpg client connection for camera 9 on port 8089 is closed
2018-12-02 04:26:57: [motioneye]  WARNING: mjpg client connection for camera 3 on port 8083 is closed
2018-12-02 04:30:57: [motioneye]  WARNING: mjpg client connection for camera 9 on port 8089 is closed
2018-12-02 04:40:17: [motioneye]  WARNING: mjpg client connection for camera 6 on port 8086 is closed
2018-12-02 04:42:33: [motioneye]    ERROR: failed to cleanup media files: [Errno 30] Read-only file system: '/data/media/sda1/office/.keep'
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/cleanup.py", line 86, in _do_cleanup
  File "/usr/lib/python2.7/site-packages/motioneye/mediafiles.py", line 293, in cleanup_media
[... 4178 lines cut ...]
 * Starting eudev: done
 * Starting watchdog: done
 * Configuring wired network: 192.168.1.100/24
 * Setting static gateway to 192.168.1.1: done
 * Setting static DNS server to 8.8.8.8: done
 * Setting current date using http: Sat Dec 1 06:41:28 CST 2018
 * Starting http date updater: done
 * Starting crond: done
 * Starting sshd: done
 * Starting proftpd: done
 * Starting smbd: done
 * Starting nmbd: done
 * Starting motioneye: done
 # Interface eth0 has IP address 192.168.1.100/24
 # Default gateway is 192.168.1.1
 # DNS server address is 8.8.8.8
---- shutting down motionEyeOS 20180627 ----
 * Stopping motioneye: done
 * Stopping smbd: done
 * Stopping nmbd: done
 * Stopping proftpd: done
 * Stopping sshd: done
 * Stopping crond: done
 * Stopping date updater: done
 * Stopping

ccrisan, 2018-12-05:
Your attached drive is read-only:
    
    2018-12-02 03:42:33: [motioneye] ERROR: failed to cleanup media files: [Errno 30] Read-only file system: '/data/media/sda1/office/.keep'

What filesystem do you use?

bsimmo, 2018-12-16:
Have the .dtb etc been updated to the newer versions? Firmware support was added (sometime after 5th September 2018)
.dtb have date 2018-11-13 and bootcode/fixup/kernel/start too.
a.k.a. is the 3A+ one included ?

As side info, I assume the HD and Ethernet are on a USB hub ?

ccrisan, 2018-12-18:
@bsimmo what file are you referring to, more precisely? All the DTBs are built from the latest Raspbian kernel.

bsimmo, 2018-12-18:
The /boot files, the dates I see are behind the latest Raspbian release and
the dates are before the 3A+ was added to the lineup.
But I'm using the stable release.
I've no idea what you testing version is using.

ccrisan, 2018-12-26:
@bsimmo the stable release does not have the latest Raspbian files. Also those file dates aren't something you should be counting on. Comparing them using `md5sum` would be the way to go.

You could upgrade to the latest prerelease; it's been some time now since I released it and no (particular) complains so far.

closed 2018-12-26 by ccrisan