motioneye-project/motioneyeos #1730: Camera stops working several times a day

thejoester, 2018-12-03:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.
Yes

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.
Yes

### motionEyeOS Version
motionEye Version | 0.39.3
Motion Version | 4.1.1
OS Version | motionEyeOS dev20181126

### Board Model

I am using the following board/model: Raspberry Pi 3 B

### Camera

I am using the following type of camera: USB Camera

My camera model is: Logitech C920

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue: Only USB Camera

### Log Files

I consider the following log files relevant to this issue:

 
[boot.log](https://github.com/ccrisan/motioneyeos/files/2639458/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/2639459/dmesg.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/2639460/messages.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/2639461/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/2639462/motioneye.log)

thejoester, 2018-12-03:
I have my RPi 3 B Running motionEyeOS dev20181126 / MotionEye 0.39.3 and USB Logitech C920 to monitor my snake. The camera will stop working several times a day to where it shows the broken camera image but the UI still loads. I can get it to restart by rebooting or by turning off/on the video source and applying but it's basically useless at this point is I have to do this as the point is to catch images and videos of my snake who often comes out when nobody is around. 

I was previously using this device to run OctoPrint to control and monitor my 3D printer, and it would run for days, even months without issue. 

I started off saving motion images and videos to a network share, but have changed it to save local - no effect,

 I was getting some errors related to power in my logs so I ordered an official RPi 2.5a power source - the errors in the logs stopped but the camera still stops working. 

I disabled the "Connectivity check" option - no effect.

I have read the other similar posts but nothing that has been suggested there has worked.

thejoester, 2018-12-04:
So tonight I replaced the older 4GB microSD card that I was using with a newer high speed Samsung 32GB microSD. I did a whole new image and we will see how this goes.

IAmOrion, 2018-12-04:
fwiw take a look here, I reported a similar issue back in July: 

github.com/ccrisan/motioneyeos/issues/1582

thejoester, 2018-12-04:
@IAmOrion - Thanks I did check that. One big difference I noticed is that I do not have to unplug the Pi, and my web GUI is still working, I have a open SSH session that does not end, and I can turn the video device off/on then apply and it starts working again.

thejoester, 2018-12-04:
OK I think I may have made some progress. 

With the new microSD card I was super hopeful as it went quite a bit longer than it had been. Then this morning it crashed again. I currently have it saving locally and not to SMB.

I also changed the still images to just save every 10 seconds, not on motion detection. One thing I noticed when I found it stopped is that the last image taken was at 09:37:10 local time. Here are the log entries from that time:

messages.log:
> Dec  4 09:37:41 Snake-Cam user.err motion: [0:motion] [ERR] [ALL] main: Thread 1 - Watchdog timeout, trying to do a graceful restart
> Dec  4 09:38:41 Snake-Cam user.err motion: [0:motion] [ERR] [ALL] main: Thread 1 - Watchdog timeout, did NOT restart graceful, killing it!
> Dec  4 09:38:42 Snake-Cam user.err motion: [0:motion] [ERR] [ALL] main: cleaning Thread 1

motion.log (last written to at 09:38)
> [0:motion] [ERR] [ALL] main: Thread 1 - Watchdog timeout, trying to do a graceful restart
> [0:motion] [ERR] [ALL] main: Thread 1 - Watchdog timeout, did NOT restart graceful, killing it!
> [0:motion] [ERR] [ALL] main: cleaning Thread 1

motioneye.log
> 2018-12-04 09:37:29: [sendmail]     INFO: sending email
> 2018-12-04 09:37:35: [sendmail]     INFO: email sent

So I have it emailing me when it detects motion, and it seems that either that, or the motion detection is causing an issue. I am disabling the notification now and see how that goes. 

I do not believe it is a network issue, as per my above reply to @IAmOrion I have an open SSH session and constant connection monitor running on my desktop and those are not erroring out.

bertdebondt, 2018-12-16:
I have the same problem as thejoester.
Every day, multiple times my Raspberry Pi3B+ loses connection with the attached raspberry pi camera module.
Log only shows that thread 1 has a watchdog timeout and could not be restarted.

[0:motion] [ERR] [ALL] main: Thread 1 - Watchdog timeout, trying to do a graceful restart
[0:motion] [ERR] [ALL] main: Thread 1 - Watchdog timeout, did NOT restart graceful, killing it!
[0:motion] [ERR] [ALL] main: cleaning Thread 1

I got this problem on my 2 rescent upgraded Pi's (refresh installed latest image (20181210) and restored my old config)

The webinterfaces is still running and a simple config change and Save button restarts the thread with again a connected camera image.

thejoester, 2018-12-17:
So two weeks later and since disabling the email notification I am not having any issues. Just to fully test this I am going to enable them.

bertdebondt, 2018-12-17:
I don't use email notification. I do use motion detection (with a fixed grid/raster).
Maybe problem is located in the motion detection module ? I will try to disable motion detection and see if camera keeps running then...

thejoester, 2018-12-17:
The last two weeks I have had motion detection enabled for movies, but not for images and have had no issues.

davisstu, 2019-01-02:
> I have the same problem as thejoester.
> Every day, multiple times my Raspberry Pi3B+ loses connection with the attached raspberry pi camera module.
> Log only shows that thread 1 has a watchdog timeout [...]

I can confirm that I am having the same issue as listed here. Pi2 with RPi Cam module. Did a fresh install on the latest release (previously had a July dev release which gave me no issues) and have both restored a config and done a fresh setup and I'm having the same issues.

It will run for a few hours, capture and upload to my Google Drive (capture on motion) then lose connection to the camera. A config change or restart brings the camera back.

bertdebondt, 2019-01-02:
It seems like more and more people have this problem running Raspberry Pi camera. Connection to the camera is lost every time motion is detected and recordings are made. Calin, did you also  experience this problem on your devellopment  system ?  I fully disabled motion recording for now and just use the camera as fast network camera option which still works. If bug can not be found I ll revert back to the working july dev release. Regards.

davisstu, 2019-01-04:
Just to confirm, with the exact same hardware and with a restoration of my config, I have rolled back to the June release and everything is working perfectly and as expected. Motion doesn't crash, I don't lose picture and motioned events are successfully uploaded to my Google Drive.

davisstu, 2019-01-05:
@ccrisan It's a good point - can you please acknowledge as an issue and we can them await a fix?

bertdebondt, 2019-01-06:
just FYI: I installed yesterday a brand new Raspbian Stretch onto a USB stick which is booted by a raspberry Pi 3B+ running the latest motion 4.2.1 and the latest MotionEye. Until now everything OK, motion is detected and recorded without interruptions or losing connection to the connected raspberry pi cam.
With other words topic problems are located in MotionEyeOS image.
Since MotionEye manually installed on Raspbian runs fine (and motion, ... packages are upgradable) I'll abandon the MotionEyeOS path for now. MotionEyeOS also can't boot from USB. So Raspbian installation is the way to go for me.
I adapted my MotionEyeOS python script to auto change some motion and motionEye settings according to night / day modus.

jasaw, 2019-01-06:
Can you guys please list step by step instructions on how to reproduce the problem? Please include detailed information on your system too, things like how many cameras, what else is attached to the RPi, configuration of your motionEyeOS (motion detection, media storage location, ...).

This may very well be a motion software bug. Try isolating the problem by running motion **without** motionEye. See https://github.com/ccrisan/motioneyeos/wiki/Report-Motion-Issue

ccrisan, 2019-01-11:
Guys, the problem of camera connection loss is something very general and may have many possible causes. Few users care to report the working and non-working versions they've tested. Even fewer of them tried to isolate the problem, as @jasaw indicated. This makes it almost impossible for us to get to the root cause of this problem (or problems).

I personally haven't experienced any similar problems with my setup (10 RPi1 FNCs + one Odroid XU4 hub). I have seen a fair amount of users complaining about this issue happening with mid 2018 versions. Some of them reported that recent prereleases (end of 2018) fixed their problems.

I will try to make some time to add Motion 4.2 support to motionEye but I have my doubts that it would fix these problems for everyone.

ccrisan, 2019-01-11:
@bertdebondt this is a really interesting finding. I never thought the OMX hardware acceleration could cause these stability issues, but now, if I think about it, it makes sense.

The last Motion update in motionEyeOS was more than 1 year ago so that couldn't really be the problem. FFmpeg on the other hand could be indeed causing the issues. @jasaw do you have any ideas how we could further debug/investigate problems that have their origin in FFmpeg RPi OMX code and result in Motion blocking or crashing?

bertdebondt, 2019-01-11:
cudo's to gui59169
Further info: on the Raspbian based motionEye ffmpeg works fine also using OMX hardware accelerated MP4 encoding. The Raspbian has 512mb allocated to the GPU (raspi-config memory split).
I think I also tried 512mb on the MotionEyeOS and it also crashed then.

ccrisan, 2019-01-11:
@bertdebondt are you sure the Raspbian version of FFmpeg comes with OMX acceleration?

ccrisan, 2019-01-11:
@bertdebondt could you please post the `ffmpeg -codecs` output from your PI, while you're at it?

bertdebondt, 2019-01-12:
> 
> 
> @bertdebondt could you please post the `ffmpeg -codecs` output from your PI, while you're at it?

ffmpeg version 3.2.10-1~deb9u1+rpt2 Copyright (c) 2000-2018 the FFmpeg developers
  built with gcc 6.3.0 (Raspbian 6.3.0-18+rpi1+deb9u1) 20170516
  configuration: --prefix=/usr --extra-version='1~deb9u1+rpt2' --toolchain=hardened --libdir=/usr/lib/arm-linux-gnueabihf --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libebur128 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmp3lame --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-omx-rpi --enable-mmal --enable-openal --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libiec61883 --arch=armhf --enable-chromaprint --enable-frei0r --enable-libopencv --enable-libx264 --enable-shared
  libavutil      55. 34.101 / 55. 34.101
  libavcodec     57. 64.101 / 57. 64.101
  libavformat    57. 56.101 / 57. 56.101
  libavdevice    57.  1.100 / 57.  1.100
  libavfilter     6. 65.100 /  6. 65.100
  libavresample   3.  1.  0 /  3.  1.  0
  libswscale      4.  2.100 /  4.  2.100
  libswresample   2.  3.100 /  2.  3.100
  libpostproc    54.  1.100 / 54.  1.100
Codecs:
 D..... = Decoding supported
 .E.... = Encoding supported
 ..V... = Video codec
 ..A... = Audio codec
 ..S... = Subtitle codec
 ...I.. = Intra frame-only codec
 ....L. = Lossy compression
 .....S = Lossless compression
 -------
 D.VI.. 012v                 Uncompressed 4:2:2 10-bit
 D.V.L. 4xm                  4X Movie
 D.VI.S 8bps                 QuickTime 8BPS video
 .EVIL. a64_multi            Multicolor charset for Commodore 64 (encoders: a64multi )
 .EVIL. a64_multi5           Multicolor charset for Commodore 64, extended with 5th color (colram) (encoders: a64multi5 )
 D.V..S aasc                 Autodesk RLE
 D.VIL. aic                  Apple Intermediate Codec
 DEVI.S alias_pix            Alias/Wavefront PIX image
 DEVIL. amv                  AMV Video
 D.V.L. anm                  Deluxe Paint Animation
 D.V.L. ansi                 ASCII/ANSI art
 DEV..S apng                 APNG (Animated Portable Network Graphics) image
 DEVIL. asv1                 ASUS V1
 DEVIL. asv2                 ASUS V2
 D.VIL. aura                 Auravision AURA
 D.VIL. aura2                Auravision Aura 2
 D.V... avrn                 Avid AVI Codec
 DEVI.. avrp                 Avid 1:1 10-bit RGB Packer
 D.V.L. avs                  AVS (Audio Video Standard) video
 DEVI.. avui                 Avid Meridien Uncompressed
 DEVI.. ayuv                 Uncompressed packed MS 4:4:4:4
 D.V.L. bethsoftvid          Bethesda VID video
 D.V.L. bfi                  Brute Force & Ignorance
 D.V.L. binkvideo            Bink video
 D.VI.. bintext              Binary text
 DEVI.S bmp                  BMP (Windows and OS/2 bitmap)
 D.V..S bmv_video            Discworld II BMV video
 D.VI.S brender_pix          BRender PIX image
 D.V.L. c93                  Interplay C93
 D.V.L. cavs                 Chinese AVS (Audio Video Standard) (AVS1-P2, JiZhun profile)
 D.V.L. cdgraphics           CD Graphics video
 D.VIL. cdxl                 Commodore CDXL video
 D.V.L. cfhd                 Cineform HD
 DEV.L. cinepak              Cinepak
 DEVIL. cljr                 Cirrus Logic AccuPak
 D.VI.S cllc                 Canopus Lossless Codec
 D.V.L. cmv                  Electronic Arts CMV video (decoders: eacmv )
 D.V... cpia                 CPiA video format
 D.V..S cscd                 CamStudio (decoders: camstudio )
 D.VIL. cyuv                 Creative YUV (CYUV)
 ..V.LS daala                Daala
 D.VILS dds                  DirectDraw Surface image decoder
 D.V.L. dfa                  Chronomaster DFA
 DEV.LS dirac                Dirac (encoders: vc2 )
[... 345 lines cut ...]
 ..D... ttf                  TrueType font
 DES... ass                  ASS (Advanced SSA) subtitle (decoders: ssa ass ) (encoders: ssa ass )
 DES... dvb_subtitle         DVB subtitles (decoders: dvbsub ) (encoders: dvbsub )
 D.S... dvb_teletext         DVB teletext (decoders: libzvbi_teletextdec )
 DES... dvd_subtitle         DVD subtitles (decoders: dvdsub ) (encoders: dvdsub )
 D.S... eia_608              EIA-608 closed captions (decoders: cc_dec )
 D.S... hdmv_pgs_subtitle    HDMV Presentation Graphic Stream subtitles (decoders: pgssub )
 ..S... hdmv_text_subtitle   HDMV Text subtitle
 D.S... jacosub              JACOsub subtitle
 D.S... microdvd             MicroDVD subtitle
 DES... mov_text             MOV text
 D.S... mpl2                 MPL2 subtitle
 D.S... pjs                  PJS (Phoenix Japanimation Society) subtitle
 D.S... realtext             RealText subtitle
 D.S... sami                 SAMI subtitle
 ..S... srt                  SubRip subtitle with embedded timing
 ..S... ssa                  SSA (SubStation Alpha) subtitle
 D.S... stl                  Spruce subtitle format
 DES... subrip               SubRip subtitle (decoders: srt subrip ) (encoders: srt subrip )
 D.S... subviewer            SubViewer subtitle
 D.S... subviewer1           SubViewer v1 subtitle
 DES... text                 raw UTF-8 text
 D.S... vplayer              VPlayer subtitle
 DES... webvtt               WebVTT subtitle
 DES... xsub                 XSUB

ccrisan, 2019-01-13:
> @ccrisan Do you remember when we switch over to buildroot that gives us buildroot change history?

Not sure what you mean, but:
 
 * 20181129 is the first version built from the git repo that has common history with BuildRoot.
 * motionEyeOS often has newer versions than BuildRoot for some important packages, such as rpi-firmware or ffmpeg

> Version 3.3.5 is good. I've been using this on all my cameras.

Maybe we should downgrade ffmpeg to `3.3.x` for a dev build and see if the problem still occurs.

> I know for a fact that rpi-firmware was broken since 1 year ago, and was only fixed in the Oct 2018 version, but we are still on the broken version.

Are you sure about that? I always update `rpi-firmware` and `rpi-userland` to latest versions used in Raspbian, at that moment. This being said, 20181209 and 20181129 should be past Oct 2018. I really hope we're talking about the prereleases here and not the latest stable 20180627.

ccrisan, 2019-01-13:
@bertdebondt your ffmpeg indeed has OMX h264 acceleration support, but as far as I know, Motion won't use it on Rasbpian, unless patched; @jasaw may know more about this.

jasaw, 2019-01-13:
Our current version of rpi-firmware (5b49caa) is from Sep 25, 2018, which still has the bug. Bug was only fixed on Oct 10, 2018. Anyway, this bug does not affect ffmpeg, so we can rule this out.

>Maybe we should downgrade ffmpeg to 3.3.x for a dev build and see if the problem still occurs.

This is exactly what I'm thinking.

As for enabling h264_omx on Raspbian, see this: https://github.com/ccrisan/motioneye/issues/930#issuecomment-402326342

ccrisan, 2019-01-15:
Guys, the [latest nightly build (dev20190114)](https://github.com/ccrisan/motioneyeos/releases/tag/nightly-dev) *should* contain the following fixes:

* ffmpeg 3.3.9
* latest rpi firmware, userland and linux, as currently used by Raspbian 

I haven't managed to test it yet though!

bertdebondt, 2019-01-15:
Hi Calin,

Just upgraded my version to your latest nightly build and activated h264 OMX movie recording again.
First I thought I ran more stable but after increasing recording motion area and more frequent recording the same problem occurs.

Log files:

motion.log:
[0:motion] [ERR] [ALL] main: Thread 1 - Watchdog timeout, trying to do a graceful restart

[0:motion] [ERR] [ALL] main: Thread 1 - Watchdog timeout, did NOT restart graceful, killing it!

[0:motion] [ERR] [ALL] main: cleaning Thread 1

motioneye.log:
2019-01-15 12:51:14: [motioneye]     INFO: updating to version dev20190114...
2019-01-15 12:51:16: [motioneye]     INFO: stopping motioneye watch script
2019-01-15 12:51:17: [motioneye]     INFO: stopping netwatch script
2019-01-15 12:51:17: [motioneye]     INFO: downloading firmware version dev20190114
2019-01-15 12:51:45: [motioneye]     INFO: extracting firmware
2019-01-15 12:52:24: [motioneye]     INFO: flashing boot partition
2019-01-15 12:52:38: [motioneye]     INFO: rebooting
2019-01-15 12:53:47: [motioneye]     INFO: hello! this is motionEye server 0.39.3
2019-01-15 12:53:50: [motioneye]     INFO: cleanup started
2019-01-15 12:53:50: [motioneye]     INFO: wsswitch started
2019-01-15 12:53:50: [motioneye]     INFO: tasks started
2019-01-15 12:53:50: [motioneye]     INFO: mjpg client garbage collector started
2019-01-15 12:53:50: [motioneye]     INFO: smb mounts started
2019-01-15 12:53:50: [motioneye]     INFO: server started
2019-01-15 12:57:02: [motioneye]     INFO: interrupt signal received, shutting down...
2019-01-15 12:57:02: [motioneye]     INFO: server stopped
2019-01-15 12:57:02: [motioneye]     INFO: tasks stopped
2019-01-15 12:57:04: [motioneye]     INFO: motion stopped
2019-01-15 12:57:04: [motioneye]     INFO: smb mounts stopped
2019-01-15 12:57:04: [motioneye]     INFO: bye!
2019-01-15 12:57:06: [motioneye]     INFO: hello! this is motionEye server 0.39.3
2019-01-15 12:57:08: [motioneye]     INFO: cleanup started
2019-01-15 12:57:08: [motioneye]     INFO: wsswitch started
2019-01-15 12:57:09: [motioneye]     INFO: tasks started
2019-01-15 12:57:09: [motioneye]     INFO: mjpg client garbage collector started
2019-01-15 12:57:09: [motioneye]     INFO: smb mounts started
2019-01-15 12:57:09: [motioneye]     INFO: server started
2019-01-15 12:58:00: [motioneye]    ERROR: failed to create movie preview for /data/media/sda1/2019-01-15/12-57-40.mp4: Command '[u'ffmpeg', u'-i', u'/data/media/sda1/2019-01-15/12-57-40.mp4', u'-f', u'mjpeg', u'-vframes', u'1', u'-ss', u'4', u'-y', u'/data/media/sda1/2019-01-15/12-57-40.mp4.thumb']' returned non-zero exit status 1
2019-01-15 13:02:40: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2019-01-15 13:02:50: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2019-01-15 13:03:00: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2019-01-15 13:03:10: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2019-01-15 13:03:20: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2019-01-15 13:03:30: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2019-01-15 13:03:40: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2019-01-15 13:03:50: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2019-01-15 13:03:59: [motioneye]  WARNING: Connect error on fd 15: ECONNREFUSED
2019-01-15 13:04:00: [motioneye]  WARNING: Connect error on fd 15: ECONNREFUSED
2019-01-15 13:04:00: [motioneye]  WARNING: Connect error on fd 15: ECONNREFUSED
2019-01-15 13:04:00: [motioneye]  WARNING: Connect error on fd 15: ECONNREFUSED
.....................................................

messages.log:
Jan 15 12:55:50 motionEye1 user.notice motion: [1994392168:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Jan 15 12:56:08 motionEye1 auth.info sshd[3027]: WARNING: could not open /data/etc/moduli (No such file or directory), using fixed modulus
Jan 15 12:56:12 motionEye1 auth.info sshd[3027]: Accepted none for admin from 192.168.1.101 port 1559 ssh2
Jan 15 12:57:00 motionEye1 cron.info crond[832]: USER root pid 3960 cmd /usr/bin/python /data/etc/sunriseset.py >> /var/log/daynight.log
Jan 15 12:57:06 motionEye1 user.notice motion: [1994281576:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
Jan 15 12:57:06 motionEye1 user.alert motion: [1994281576:motion] [ALR] [ALL] conf_cmdparse: Deprecated config option "thread" since after version 3.4.1:
Jan 15 12:57:06 motionEye1 user.alert motion: [1994281576:motion] [ALR] [ALL] conf_cmdparse: The "thread" option has been replaced by the "camera" option.
Jan 15 12:57:06 motionEye1 user.notice motion: [1994281576:motion] [NTC] [ALL] config_camera: Processing camera config file thread-1.conf
Jan 15 12:57:06 motionEye1 user.notice motion: [1994281576:motion] [NTC] [ALL] motion_startup: Motion 4.1.1 Started
Jan 15 12:57:06 motionEye1 user.notice motion: [1994281576:motion] [NTC] [ALL] motion_startup: Logging to syslog
Jan 15 12:57:06 motionEye1 user.notice motion: [1994281576:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
Jan 15 12:57:06 motionEye1 user.notice motion: [1994281576:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[... 261 lines cut ...]
[... cut ...]
[Thu Jan  1 01:00:06 1970] urandom_read: 2 callbacks suppressed
[Thu Jan  1 01:00:06 1970] random: ssh-keygen: uninitialized urandom read (32 bytes read)
[Thu Jan  1 01:00:08 1970] random: sshd: uninitialized urandom read (32 bytes read)
[Thu Jan  1 01:00:08 1970] random: python: uninitialized urandom read (2500 bytes read)
[Thu Jan  1 01:00:09 1970] random: python: uninitialized urandom read (32 bytes read)
[Thu Jan  1 01:00:09 1970] random: python: uninitialized urandom read (32 bytes read)
[Thu Jan  1 01:00:22 1970] random: crng init done

ccrisan, 2019-01-15:
@bertdebondt that's a bummer. So ffmpeg and rpi-firmware are not to blame here. Guys, any more ideas?

davisstu, 2019-01-15:
What was the reason for not upgrading to the latest version of Motion? It looks like MotionEye uses an old version of Motion, surely that's the next obvious step - to upgrade to the latest version?

ccrisan, 2019-01-15:
@davisstu motionEye doesn't work well with Motion 4.2.

jasaw, 2019-01-16:
@ccrisan @bertdebondt I managed to reproduce the problem at my end thanks to another user who is experiencing the same problem and documented steps to trigger the bug. The issue comes from either rpi-firmware or rpi-userland. This bug only shows up when the GPU is unable to cope with the data and motion starts reporting timeout on the camera thread (thread-1 in our case) and starts killing the thread. The logs below is what motion reports when we encounter this problem. Users who encountered this issue have loaded up the GPU either by pushing high resolution, or by pushing multiple camera streams to the GPU to encode.

>[1:ml1] [INF] [EVT] event_new_video: Source FPS 2
[1:ml1] [NTC] [ENC] ffmpeg_set_codec: Using codec h264_omx
[1:ml1] [INF] [ENC] ffmpeg_set_quality: h264_omx codec vbr/crf/bit_rate: 2250000
[1:ml1] [NTC] [EVT] event_newfile: File of type 8 saved to: /data/output/TestCam/2019-01-16/12-00-31.mp4
[1:ml1] [NTC] [ALL] motion_detected: Motion detected - starting event 1
[1:ml1] [DBG] [ENC] ffmpeg_encode_video: Receive packet threw EAGAIN returning -2 code :Resource temporarily unavailable
[1:ml1] [DBG] [ENC] ffmpeg_put_image: Buffered packet
[1:ml1] [DBG] [ENC] ffmpeg_encode_video: Receive packet threw EAGAIN returning -2 code :Resource temporarily unavailable
[1:ml1] [DBG] [ENC] ffmpeg_put_image: Buffered packet
[1:ml1] [DBG] [ENC] ffmpeg_encode_video: Receive packet threw EAGAIN returning -2 code :Resource temporarily unavailable
[1:ml1] [DBG] [ENC] ffmpeg_put_image: Buffered packet
[1:ml1] [DBG] [ENC] ffmpeg_encode_video: Receive packet threw EAGAIN returning -2 code :Resource temporarily unavailable
[1:ml1] [DBG] [ENC] ffmpeg_put_image: Buffered packet
[1:ml1] [DBG] [ENC] ffmpeg_encode_video: Receive packet threw EAGAIN returning -2 code :Resource temporarily unavailable
[1:ml1] [DBG] [ENC] ffmpeg_put_image: Buffered packet
[0:motion] [ERR] [ALL] main: Thread 1 - Watchdog timeout, trying to do a graceful restart
[0:motion] [ERR] [ALL] main: Thread 1 - Watchdog timeout, did NOT restart graceful, killing it!
[0:motion] [ERR] [ALL] main: cleaning Thread 1

I rolled the rpi-firmware and rpi-userland back to the version used in 20180627, and this problem disappeared. I don't know what change triggered this bug in the rpi-firmware/userland. https://github.com/raspberrypi/firmware/issues/1051#issuecomment-426232817 might have something to do with it?

What can we do to address this issue? I don't know yet...

ccrisan, 2019-01-16:
@jasaw that's a very good finding. For now, I'll downgrade the rpi-firmware and userland and hope they work well with the latest kernel & everything.

In the meantime, since the issue you referenced in rpi-firmware seems to have been fixed for everyone, I believe it's not our issue. Nevertheless, it may be worth reporting our problem to them.

Do you think Motion-4.2 has a chance to fix/workaround this problem?

jasaw, 2019-01-17:
@ccrisan After a bit more testing, I'm starting to think this is a motion software problem. When motion prints "Buffered packet" and EAGAIN, it is still recording the video just fine. After 30 seconds of EAGAIN messages, motion decided that watchdog has timed out and kills the camera thread even though the camera thread is still doing its job. I'm marking this as a motion bug for now. I've reported this to motion project too: https://github.com/Motion-Project/motion/issues/888

I don't know what the side effects are for using old rpi-firmware/userland with new kernel, but we should still roll back as a temporary workaround until we can fix it in motion.

Regarding motion 4.2, I highly doubt that 4.2 is going to fix this problem.

ccrisan, 2019-01-17:
@jasaw I will happily apply the patch myself but I'm a bit lost here. The rpi-firmware bug 1087 you mention, is that *the* bug? So you're not suspecting Motion anymore?

A new nightly build will be soon available with `rpi-firmware` and `rpi-userland` from summer 2018, hoping it'll make this problem go away.

Again, if you think or know that the patch solves this problem entirely, even for recent `rpi-firmware` variants, do let me know so I can upgrade the packages back again and get this fixed for once and for always.

6by9, 2019-01-17:
(Pi firmware developer here)
I can't say whether raspberrypi/firmware#1087 is your actual bug, but to give some history.

FFmpeg appears to have always had an issue in how it handles OMX components.
IL allows you to split encoded frames over multiple buffers if required. FFmpeg attempts to handle this by checking for the ```OMX_BUFFERFLAG_ENDOFFRAME``` flag, but it then doesn't wait for a buffer to be returned.
The default configuration is for only a single output buffer, so the likelihood of having just returned that one buffer to the GPU and expecting it to be filled and returned by a dozen lines later is incredibly low. The default buffer size is 64kB, so handles the majority of frames except at very high bitrates.

In the firmware release of 9th July we changed the video_encode component to use a hardware block in the input stage as it gives a significant performance advantage. This tweaks the timing slightly, and FFmpeg's flawed output buffer handling starts failing to collect all the buffers in a frame. The codec output FIFO (2MB) isn't being drained correctly, and eventually the codec stalls due to having no space remaining.

The diff to FFmpeg corrects the behaviour so that it collects all the data in each frame, and therefore the codec won't stall.

jasaw, 2019-01-17:
@6by9 Thank you for the detailed explanation. Your ffmpeg `OMX_BUFFERFLAG_ENDOFFRAME` patch has been working well for me.

@ccrisan The bug appears to be in ffmpeg and not motion software. I've created a PR that adds this `OMX_BUFFERFLAG_ENDOFFRAME` patch, and we should be able to revert back to latest rpi-firmware.

ccrisan, 2019-01-17:
Guys, that's great news. I'll make sure to prepare the next nightly build according to all these, but unfortunately my CI machine just crashed and I need to see what's going on. I'll keep you posted, here.

ccrisan, 2019-01-18:
[dev20190118](https://github.com/ccrisan/motioneyeos/releases/tag/nightly-dev) *should* contain the ffmpeg patch added by @jasaw. Please test it and let us know if the problem is fixed.

rullywowr, 2019-01-18:
Thank you all. I installed the new dev20190118 build at 1:30 UTC (8:30AM EST). Will report back in 24h to provide an update on the status. Glad to provide any additional info you may like to see.

bertdebondt, 2019-01-18:
I also installed it and it looks like the problem is fixed.
Raspberry Pi OMX MP4 h264 encoding is working now without losing Pi camera connection. Will also report back in full 24h for final update. But it lookd good. Made already motion recording where in previous situation, recording already crashed.

ccrisan, 2019-01-19:
Closed via b80aedfeaaa421ba4fbfea4afdc8ae4a4fd1f11a.

closed 2019-01-19 by ccrisan

rullywowr, 2019-01-20:
Fixed for me too! Thank you all.

IAmOrion, 2019-01-23:
If I set up the new nightly build, can subsequent versions be updated ok? (I'm assuming the warning about not updating is only if you're coming from 20180627 or earlier?)

davisstu, 2019-01-23:
> If I set up the new nightly build, can subsequent versions be updated ok? (I'm assuming the warning about not updating is only if you're coming from 20180627 or earlier?)

Worst case scenario would be to backup your config file, fresh install and reapply config. I tend to do this on major releases anyway, as the inbuilt upgrade tends to be unreliable.

ccrisan, 2019-01-23:
@IAmOrion yes, you will be able to upgrade seamlessly afterwards. Nevertheless I recommend using the latest pre-release (instead of nightly). No significant difference, just the fact that it's a proper release with version & everything.

IAmOrion, 2019-01-23:
> Worst case scenario would be to backup your config file, fresh install and reapply config. I tend to do this on major releases anyway, as the inbuilt upgrade tends to be unreliable.

The problem with that, is in my usage case, my Pi's are all remote and a royal pain the behind to get to.

> @IAmOrion yes, you will be able to upgrade seamlessly afterwards. Nevertheless I recommend using the latest pre-release (instead of nightly). No significant difference, just the fact that it's a proper release with version & everything.

Great, thanks.  I did actually mean the pre-release (20190119) and not nightly, but thanks for the info.

I'm hoping this will also fix my issues from here https://github.com/ccrisan/motioneyeos/issues/1582

functionalprototype, 2020-11-26:
I'm seeing this problem on a Raspberry Pi 3 Model B Plus Rev 1.3, motionEye Version | 0.42.1, Motion Version | 4.2.2+gitUNKNOWN, OS Version | motionEyeOS 20200606

Should I open a new issue or ask that this one be re-opened?

starbasessd, 2020-11-26:
Open a new on if 20190911, 20200203, or dev20201026 do not resolve it. There are many known issues with 20200606.