motioneye-project/motioneyeos #1733: SSH PORT CHANGE

RottenDirector, 2018-12-04:
So I have 2 ssh computers using ssh one on my main pc and one on my motioneye rasp pi I can access it using ssh when on my home network using its local ip but when trying to connect to it outside my home network using my external ip but it defaults to my pc how do I change the port from 22 to 23 on my Pi can someone tell me how in commands as I’m a complete noob with command lines

jasaw, 2018-12-04:
You don't need to change the SSH port number on your Pi. You have to configure your router to:
1. forward external port 22 to your PC port 22.
2. forward external port 23 to your Pi port 22.

tigernero79, 2018-12-05:
Via ssh digit :
1) mount -o remount,rw /
2) nano /etc/sshd_config
3) change voice port ssh 22 then another bomber
4) mount -o remount,ro /
5) reboot
The end