motioneye-project/motioneyeos #1734: USB2 host and USB3 host is not working with OPI-ONE

yoshibatora, 2018-12-07:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motioneyeos-orangepione-20181129.

### Board Model

I am using the OrangePi One.

### Network Connection

My motionEyeOS unit is connected to the network via WiFi, and the USB2 host and USB3 host is not enabled, I cannot found my Wifi card using lsusb. I can enable USB2 and USB3 hosts using Armbian, run the armbian-config script and choosing hardware-USB2 host and USB3 host option, but not motioneyeOS.

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

Rtl8188EU USB WiFi adapter, but the driver is in module folder.

### Log Files

I consider the following log files relevant to this issue:

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.1 (root@ce2485dbc5a4) (gcc version 7.3.1 20180425 [linaro-7.3-2018.05 revision d29120a424ecfbc167ef90065c0eeb7f91977701] (Linaro GCC 7.3-2018.05)) #1 SMP Sun Dec 2 23:18:45 UTC 2018
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=30c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Xunlong Orange Pi One
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 16 MiB at 0x000000005f000000
[    0.000000] On node 0 totalpages: 131072
[    0.000000]   Normal zone: 1152 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 131072 pages, LIFO batch:31
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: Using PSCI v0.1 Function IDs from DT
[    0.000000] random: get_random_bytes called from start_kernel+0xa0/0x440 with crng_init=0
[    0.000000] percpu: Embedded 17 pages/cpu @(ptrval) s37068 r8192 d24372 u69632
[    0.000000] pcpu-alloc: s37068 r8192 d24372 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129920
[    0.000000] Kernel command line: earlyprintk console=tty1 root=/dev/mmcblk0p2 rootwait ro no_console_suspend panic=10 quiet loglevel=1 ipv6.disable=1
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 483084K/524288K available (12288K kernel code, 506K rwdata, 2644K rodata, 2048K init, 438K bss, 24820K reserved, 16384K cma-reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
                   lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
                   pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
                   modules : 0xbf000000 - 0xbfe00000   (  14 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (14304 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   (2048 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 507 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 439 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=4.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 79635851949 ns
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[... 328 lines cut ...]
[    2.002160] vcc5v0: disabling
[    2.002164] usb0-vbus: disabling
[    2.002172] ALSA device list:
[    2.002176]   #0: Dummy 1
[    2.002179]   #1: Loopback 1
[    2.003006] platform regulatory.0: Direct firmware load for regulatory.db failed with error -2
[    2.003014] cfg80211: failed to load regulatory.db
[    2.010155] EXT4-fs (mmcblk0p2): mounted filesystem without journal. Opts: (null)
[    2.010204] VFS: Mounted root (ext4 filesystem) readonly on device 179:2.
[    2.011205] devtmpfs: mounted
[    2.013349] Freeing unused kernel memory: 2048K
[    2.025171] Run /sbin/init as init process
[    2.577969] random: fast init done
[    2.739992] [drm] Cannot find any crtc or sizes
[   35.500085] random: crng init done
[   37.165402] EXT4-fs (mmcblk0p3): mounted filesystem with ordered data mode. Opts: (null)
[   39.063414] udevd[330]: starting version 3.2.5
[   39.137735] udevd[331]: starting eudev-3.2.5
[   39.391663] asoc-simple-card soc:sound: i2s-hifi <-> 1c22800.i2s mapping ok
[   39.943273] Generic PHY 0.1:01: attached PHY driver [Generic PHY] (mii_bus:phy_addr=0.1:01, irq=POLL)
[   39.945560] dwmac-sun8i 1c30000.ethernet eth0: No Safety Features support found
[   39.945574] dwmac-sun8i 1c30000.ethernet eth0: No MAC Management Counters available
[   39.945583] dwmac-sun8i 1c30000.ethernet eth0: PTP not supported by HW
[   42.020541] dwmac-sun8i 1c30000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[   66.854432] system date is 1970-01-01 00:01:12

yoshibatora, 2018-12-07:
Also found these two lines of error.
platform regulatory.0: Direct firmware load for regulatory.db failed with error -2
cfg80211: failed to load regulatory.db

ccrisan, 2018-12-15:
@yoshibatora I understand your problem and appreciate the effort you put in providing the log files but unfortunately I can't help you with this issue. I'm simply compiling the "official" kernel with their default configuration.

If anyone is able to spot the problem here, please feel free to suggest improvements to the Orange PI One build.