motioneye-project/motioneyeos #1735: 20181129 Pre Release

duckietm, 2018-12-07:
Hello,

I read that we can't do an Live Update in the 20181129 Pre Release.
Will this fixed in the stable release?

ccrisan, 2018-12-15:
@duckietm most likely not, unless someone jumps in and actually implements an intelligent and safe mechanism that deals with the partition layout change in a non-destructive way.

@jasaw proposed a solution but I can't seem to find the issue where we've been discussing this.

jasaw, 2018-12-16:
@ccrisan @duckietm The solution that I proposed is here: https://github.com/ccrisan/motioneyeos/issues/1582#issuecomment-443921121

duckietm, 2018-12-17:
Yes, agree completely. We could add a checkbox next to firmware upgrade button that says "reformat data partition". We could also add a warning about losing media files. I think that should satisfy a lot of users.

This would be a great option, this will save me alot of work !

jasaw, 2018-12-29:
Created "reformat data partition" issue here: https://github.com/ccrisan/motioneyeos/issues/1756

closed 2019-01-02 by ccrisan